adafruit/Adafruit_Python_PlatformDetect #20: BeagleBoards - root privileges are required for reading board id

s-light, 2019-03-28:
currently the call to `with open("/sys/bus/nvmem/devices/0-00500/nvmem", "rb") as eeprom:`
requires root privileges.
this leads to scripts that use Blinka and are run on a BeagleBoard require root (sudo) execution.

this seems to be a known limitation regarding beagle eeprom reading..
one relevant discussion:
https://groups.google.com/d/msg/beagleboard/rtB2f4Aqd-Q/I8GCpvTlAAAJ

after some searching i found [this blog post](https://martin-jones.com/2016/08/12/making-beaglebone-black-serial-number-available-to-user-applications/) that describes how to add a udev rule to allow reading the eeprom from the i2c group.

i have extended this example to also allow this for the nvmem path. its working nicely :-)
you find it attached.

would it make sens to include it in this library?
and eventually install it on lib-install?
(i have no idea if this is a good idea..)

[80-eeprom-noroot.rules.txt](https://github.com/adafruit/Adafruit_Python_PlatformDetect/files/3020228/80-eeprom-noroot.rules.txt)

ladyada, 2019-03-28:
i think we should error if they try to run it and cannot read the eeprom, and then point them to the guide on how to add the udev!

pdp7, 2019-03-30:
An alternative to the nvmem file is `/proc/device-tree/model` which is would readable and provides the necessary information.

For example, Adafruit_BBIO uses this to determine the special case of PocketBeagle (which has P1/P2 headers instead of P8/P9 headers):
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L578
```
    const char *cmd = "/bin/grep -c 'TI AM335x PocketBeagle' /proc/device-tree/model";
```
and Blue which has specific JST connectors:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L606
```
    const char *cmd = "/bin/grep -c 'TI AM335x BeagleBone Blue' /proc/device-tree/model";
```

@robertcnelson Is there a list of all the `/proc/device-tree/model` values?  

@brennen compiled the eeprom strings [here](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/board.py#L99) and I think something similar would be needed for the `/proc/device-tree/model` string

RobertCNelson, 2019-03-30:
@pdp7  i have a generated list here:

https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-4.14.y/patches/all.db

```
Machine: TI AM335x BeagleBone
Machine: TI AM335x BeagleBone Black
Machine: TI AM335x BeagleBone Black RoboticsCape
Machine: TI AM335x BeagleBone Black Wireless
Machine: TI AM335x BeagleBone Black Wireless RoboticsCape
Machine: TI AM335x BeagleBone Blue
Machine: TI AM335x BeagleBone Green
Machine: TI AM335x BeagleBone Green Wireless
Machine: Arrow BeagleBone Black Industrial
Machine: BeagleBoard.org BeagleBone AI
Machine: SanCloud BeagleBone Enhanced
```

ps, please drop the nvmem path so if you have to use eeprom, just use "/sys/bus/i2c/devices/0-0050/eeprom"

The nvmem path was only from 4.1.x-4.4.x, after that mainline decided nvmem should be compatible with old eeprom path...

Regards,

pdp7, 2019-03-31:
@RobertCNelson thanks for the list.  I think this will be very useful if this library switches to using `/proc/device-tree/model`.

For `/sys/bus/i2c/devices/0-0050/eeprom`, have you considered adding udev rule to the Debian image to allow the `debian` user to read that file?

s-light, 2019-04-01:
so if i understand this correct - the prefered method of receiving the Board Type would be to look at
`/proc/device-tree/model`

currently there is an own list / mapping from raw board-ids to board names at [adafruit_platformdetect/board.py#L102](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/board.py#L102)
it is used in [`def _beaglebone_id(self)`adafruit_platformdetect/board.py#L301](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/board.py#L301)
- this function returns the board model as string.
so here we could change to the `proc/device-tree/model` reading - eventually - for backwards-compatibility - have to remap this to the internal already used model-names. and are done?

i think i could do this - if not someone else is on it already...

pdp7, 2019-04-02:
fyi - @RobertCNelson informed me in the beagleboard.org slack that even with the correct udev rules, a non-root user can still not read /sys/bus/i2c/devices/0-0050/eeprom

Thus, I think the only solution to this issue is to use `/proc/device-tree/model`

@s-light I don't think backwards compatibility is necessary as I think the detection of Beagle based boards needs to be refactored.  This would effect:
https://github.com/adafruit/Adafruit_Blinka/blob/master/src/board.py
and
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/board.py

The main difference between Beagle-based boards for this library is the headers.  There are 3 groups:
* BeagleBone-based boards with P8/P9 headers
* BeagleBone Blue with specific JST connectors
* PocketBeagle with P1/P2 headers

This is the list of possible board types in `/proc/device-tree/model`:
https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-4.19.y/patches/all.db

The first step is to take the Beagle-based boards and put them into one of the 3 categories I listed above.

RobertCNelson, 2019-04-02:
Correction, my knowledge of udev wasn't enough to formulate a working udev rule.. ;)

I'm sure it could be done.

Regards,

```code
debian@test-bbb-2:~$ sudo udevadm info -a /sys/bus/i2c/devices/0-0050 

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/platform/ocp/44e0b000.i2c/i2c-0/0-0050':
    KERNEL=="0-0050"
    SUBSYSTEM=="i2c"
    DRIVER=="at24"
    ATTR{name}=="24c256"

  looking at parent device '/devices/platform/ocp/44e0b000.i2c/i2c-0':
    KERNELS=="i2c-0"
    SUBSYSTEMS=="i2c"
    DRIVERS==""
    ATTRS{name}=="OMAP I2C adapter"

  looking at parent device '/devices/platform/ocp/44e0b000.i2c':
    KERNELS=="44e0b000.i2c"
    SUBSYSTEMS=="platform"
    DRIVERS=="omap_i2c"
    ATTRS{driver_override}=="(null)"

  looking at parent device '/devices/platform/ocp':
    KERNELS=="ocp"
    SUBSYSTEMS=="platform"
    DRIVERS==""
    ATTRS{driver_override}=="(null)"

  looking at parent device '/devices/platform':
    KERNELS=="platform"
    SUBSYSTEMS==""
    DRIVERS==""
```

```code
debian@beaglebone:~$ cat /etc/udev/rules.d/51-i2c.rules
#KERNEL=="i2c-[0-7]*", GROUP="i2c", MODE="0660"

SUBSYSTEM=="i2c", DRIVER=="at24", GROUP="i2c", MODE="0660
```

pdp7, 2019-04-02:
@RobertCNelson thanks for clarifying

pdp7, 2019-04-03:
@RobertCNelson do you think there is any downside to using `/proc/device-tree/model` to identify the type of Beagle?

pdp7, 2019-04-03:
@RobertCNelson Could [80-eeprom-noroot.rules](https://github.com/s-light/pocketbeagle_python_tests/blob/master/80-eeprom-noroot.rules) by @s-light be added to the BeagleBoard.org Debian image?

i just tested this on a PocketBeagle and it worked okay.

```
debian@beaglebone:/etc/udev/rules.d$ sudo wget https://raw.githubusercontent.com/s-light/pocketbeagle_python_tests/master/80-eeprom-noroot.rules^C
debian@beaglebone:/etc/udev/rules.d$ cat ./80-eeprom-noroot.rules 
# /etc/udev/rules.d/80-eeprom-noroot.rules
#
# allow i2c group to read of eeprom
# based on:
# https://martin-jones.com/2016/08/12/making-beaglebone-black-serial-number-available-to-user-applications/
# extended by: Stefan Krüger
#
# allows both locations to be read by the i2c group:
# /sys/bus/i2c/devices/0-0050/eeprom
# /sys/bus/nvmem/devices/0-00500/nvmem
#
# check result:
# ls -l /sys/bus/i2c/devices/0-0050/eeprom /sys/bus/nvmem/devices/0-00500/nvmem
# both should be readable. similar to this
# -rw-r----- 1 root i2c 0 XXXXXXXXX
#

SUBSYSTEM=="i2c", DEVPATH=="*0-0050", RUN+="/bin/chown root:i2c /sys$env{DEVPATH}/eeprom", RUN+="/bin/chmod 0640 /sys$env{DEVPATH}/eeprom"
SUBSYSTEM=="nvmem", DEVPATH=="*0-00500", RUN+="/bin/chown root:i2c /sys$env{DEVPATH}/nvmem", RUN+="/bin/chmod 0640 /sys$env{DEVPATH}/nvmem"
debian@beaglebone:/etc/udev/rules.d$ ls -ltar /sys/bus/i2c/devices/0-0050/eeprom
-rw-r----- 1 root i2c 32768 Apr  3 11:52 /sys/bus/i2c/devices/0-0050/eeprom
debian@beaglebone:/etc/udev/rules.d$ hexdump -C /sys/bus/i2c/devices/0-0050/eeprom |head
00000000  aa 55 33 ee 41 33 33 35  50 42 47 4c 30 30 41 32  |.U3.A335PBGL00A2|
00000010  31 38 33 34 47 50 42 35  30 32 32 34 ff ff ff ff  |1834GPB50224....|
00000020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00001000  aa 55 33 ee 41 33 33 35  50 42 47 4c 30 30 41 32  |.U3.A335PBGL00A2|
00001010  31 38 33 34 47 50 42 35  30 32 32 34 ff ff ff ff  |1834GPB50224....|
00001020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00002000  aa 55 33 ee 41 33 33 35  50 42 47 4c 30 30 41 32  |.U3.A335PBGL00A2|
00002010  31 38 33 34 47 50 42 35  30 32 32 34 ff ff ff ff  |1834GPB50224....|
debian@beaglebone:/etc/udev/rules.d$ cd ~/pocketbeagle_python_tests/
debian@beaglebone:~/pocketbeagle_python_tests$ python3 ./blinka_test.py 
Hello blinka!
Digital IO ok!
I2C ok!
I2C_2 ok!
SPI ok!
SPI_1 ok!
done!
debian@beaglebone:~/pocketbeagle_python_tests$ sudo /opt/scripts/tools/version.sh 
[sudo] password for debian: 
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335PBGL00A21834GPB50224]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.080162] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.088390] gpio-of-helper ocp:cape-universal: ready
END
```

pdp7, 2019-04-03:
@s-light I got an update from @RobertCNelson in our BeagleBoard.org slack (let me know if you want an invite: drew@beagleboard.org) and the udev rule that you created will be in the next Debian image (usually there are new images every week or so on https://elinux.org/Beagleboard:BeagleBoneBlack_Debian )

RobertCNelson, 2019-04-03:
Today just run:
```
sudo apt update
sudo apt install --only-upgrade bb-customizations
sudo reboot
```

pdp7, 2019-04-04:
@RobertCNelson thanks!

@s-light for reference, this is the commit that added your udev rule for i2c nvmem:
https://github.com/rcn-ee/repos/commit/3efd567a81bbdd262a50d6167f93c0e7765379cf

pdp7, 2019-04-05:
@RobertCNelson thanks!

I ran:
```
sudo apt install --only-upgrade bb-customizations
sudo /opt/scripts/tools/update_kernel.sh 
```
after reboot, I have the new /dev/spi/X.Y symlinks:
```
debian@beaglebone:~$ ls -ltar /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr  5 14:09 /dev/spidev1.0
crw-rw---- 1 root spi 153, 1 Apr  5 14:09 /dev/spidev1.1
crw-rw---- 1 root spi 153, 3 Apr  5 14:09 /dev/spidev2.1
crw-rw---- 1 root spi 153, 2 Apr  5 14:09 /dev/spidev2.0
```
```
debian@beaglebone:~$ ls -ltar /dev/spi/
total 0
drwxr-xr-x  2 root root  120 Nov  3  2016 .
drwxr-xr-x 16 root root 3320 Apr  5 14:09 ..
lrwxrwxrwx  1 root root   12 Apr  5 14:09 0.0 -> ../spidev1.0
lrwxrwxrwx  1 root root   12 Apr  5 14:09 0.1 -> ../spidev1.1
lrwxrwxrwx  1 root root   12 Apr  5 14:09 1.1 -> ../spidev2.1
lrwxrwxrwx  1 root root   12 Apr  5 14:09 1.0 -> ../spidev2.0
```
```
debian@beaglebone:~$ uname -r
4.14.108-ti-r103
debian@beaglebone:~$ cat /etc/dogtag 
BeagleBoard.org Debian Image 2019-03-03
debian@beaglebone:~$ cat /etc/debian_version 
9.8
```
```
debian@beaglebone:~$ cat /etc/udev/rules.d/10-of-symlink.rules 
# vim: ft=udevrules

#From: https://github.com/mvduin/py-uio/blob/master/etc/udev/rules.d/10-of-symlink.rules

# allow declaring a symlink for a device in DT

ATTR{device/of_node/symlink}!="", \
	ENV{OF_SYMLINK}="%s{device/of_node/symlink}"

ENV{OF_SYMLINK}!="", ENV{DEVNAME}!="", \
	SYMLINK+="%E{OF_SYMLINK}", \
	TAG+="systemd", ENV{SYSTEMD_ALIAS}+="/dev/%E{OF_SYMLINK}"
debian@beaglebone:~$ 
```

makermelissa, 2019-05-31:
@s-light, it's been awhile. Is this still an issue?

s-light, 2019-06-06:
sorry for the silence from my side... 
i hope to get back to this at the weekend/beginning next week

s-light, 2019-06-21:
sorry for the long delay..

preparations before the command as in https://github.com/adafruit/Adafruit_Blinka/issues/100#issuecomment-504588038
(fresh image on card + apt update + pip3 install adafruit-circuitpython-xxx lib..)

```bash
debian@beaglebone:~$ ls -ltar /sys/bus/i2c/devices/0-0050/eeprom
-rw-r----- 1 root i2c 32768 Jun 21 19:55 /sys/bus/i2c/devices/0-0050/eeprom
debian@beaglebone:~$ hexdump -C /sys/bus/i2c/devices/0-0050/eeprom |head  -3
00000000  aa 55 33 ee 41 33 33 35  50 42 47 4c 30 30 41 32  |.U3.A335PBGL00A2|
00000010  31 37 34 33 47 50 42 32  30 34 35 37 ff ff ff ff  |1743GPB20457....|
00000020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
```

seems to be all fine!
i think we can close this.

makermelissa, 2019-06-21:
Awesome! Glad to hear it’s working.

closed 2019-06-21 by makermelissa