adafruit/Adafruit_Python_PlatformDetect #45: Possibility to read info from OpenWRT devices

yeyeto2788, 2019-12-20:
Hello guys! 

I'm trying to get all info to add the Onion Omega (first version) into the project, digging a bit through the code I see that you're using a regex looking in the `/proc/cpuinfo` so it is possible to add the board into the project so it detects the board as the Onion Omega or at least as generic OpenWRT device?

How can I proceed with it?

Just in case, here you have some info that might help with it:
```console
root@Omega-198C:/# uname -a
Linux Omega-198C 4.9.152 #0 Mon Jan 28 08:54:32 2019 mips GNU/Linux

root@Omega-198C:/# cat /proc/cpuinfo 
system type             : Atheros AR9330 rev 1
machine                 : Onion Omega
processor               : 0
cpu model               : MIPS 24Kc V7.4
BogoMIPS                : 265.42
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 16
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available
```

Regards.

ladyada, 2019-12-20:
you can use whatever you like, maybe 'machine' and 'system type' would be more useful for you? please submit a PR for whatever hardware detection you need :)

yeyeto2788, 2019-12-21:
Hey @ladyada,

Ok, I'll make a PR with your proposal.

After the detection is done, can you please point me out to what are the next steps into making blinka available for this board?

Regards.

ladyada, 2019-12-21:
@makermelissa can help you out :)

yeyeto2788, 2019-12-23:
@ladyada thanks for the reply! @makermelissa I hope I did it right. If there is anything else I can do just let me know.

Regards.

makermelissa, 2019-12-30:
Hi, I commented on the PR. I just noticed in your summary there is a system type field that has the AR9330 or the cpu model of MIPS 24Kc. If we could use either of those for the chip, that would be excellent.

yeyeto2788, 2019-12-30:
Hey @makermelissa thanks for taking the time on reviewing the PR. That sounds great, I'll make the changes to the PR regarding this.

Regards.

makermelissa, 2020-01-03:
Ok, this has now been merged into PlatformDetect. Here's my cpuinfo dump off the Omega2 for completeness:
```
root@OpenWrt:~# cat /proc/cpuinfo
system type             : MediaTek MT7688 ver:1 eco:2
machine                 : Onion Omega2+
processor               : 0
cpu model               : MIPS 24KEc V5.5
BogoMIPS                : 385.84
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16 dsp
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available
```

closed 2020-01-03 by makermelissa