adafruit/Adafruit_Python_PlatformDetect #49: Detection of Pine64 boards/devices

yeyeto2788, 2020-01-07:
As per issue https://github.com/adafruit/Adafruit_Blinka/issues/118, I'm creating this issue in order to keep track the changes regarding the addition of Pine64 boards/devices.

@makermelissa I'll need the output of the `/proc/cpuinfo` from your board so I can add the different boards/devices from Pine64.

Should I also add the PineBook and the PinePhone?

Regards.

yeyeto2788, 2020-01-07:
PR #52 already added but still missing more accurate details from files:
`/proc/cpuinfo` and the `/proc/device-tree/model`

makermelissa, 2020-01-09:
Ok, got the output of `/proc/cpuinfo` here for the Pine A64:
```
pi@pine64:~$ cat /proc/cpuinfo
processor	: 0
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
```

yeyeto2788, 2020-01-09:
Hey @makermelissa , thanks for the info.

I have just changed the code in order to detect the board. Can you please test it out on your board?

Regards.

makermelissa, 2020-01-09:
Of course!

makermelissa, 2020-01-09:
Fixed by #52

closed 2020-01-09 by makermelissa