adafruit/Adafruit_Python_PlatformDetect #55: Possible library refactor

yeyeto2788, 2020-01-20:
Hey @ladyada, @makermelissa,

I have done a little refactor on the library so when it becomes bigger detecting more boards we don't have 10k lines modules within the library.

I have made the changes here https://github.com/yeyeto2788/Adafruit_Python_PlatformDetect/tree/module_refactor, if you wish I can make a PR with this changes.

It still works as before to save retro-compatibility issue in case there is any:
```console
import adafruit_platformdetect
detector = adafruit_platformdetect.Detector()
detector.board.id
'GENERIC_LINUX_PC'
detector.chip.id
'GENERIC_X86'
detector.board.any_embedded_linux
False
```

Regards.

ladyada, 2020-01-20:
yah plz make a PR, it looks good!

yeyeto2788, 2020-01-20:
Great! Done.

yeyeto2788, 2020-01-20:
Ups! Build failing :( due to naming issues. I'll fix that once I get home.

yeyeto2788, 2020-01-20:
Fixed :P

yeyeto2788, 2020-01-22:
We can close this issue when PR https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/59 is merged and Blinka is finally working.

closed 2020-01-22 by yeyeto2788