adafruit/Adafruit_Python_PlatformDetect #57: failure to detect BeagleBone Green

pdp7, 2020-01-21:
The 1st generation BeagleBone Green has an unusual board ID:

Here is the SeeedStudio BeagleBone Green [from image-builder README](https://github.com/beagleboard/image-builder):
```
  1A: [aa 55 33 ee 41 33 33 35  42 4e 4c 54 1a 00 00 00 |.U3.A335BNLT....|]
   ?: [aa 55 33 ee 41 33 33 35  42 4e 4c 54 42 42 47 31 |.U3.A335BNLTBBG1|]
```

[adafruit_platformdetect/board.py](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/board.py#L195) tries to handle this with:
```
    BEAGLEBONE_GREEN: (
        ('1A', 'A335BNLT....'),
        ('UNKNOWN', 'A335BNLTBBG1'),
    ),
```

However, it should be `A335BNLT` followed by 0`x1a`:
```
41 33 33 35  42 4e 4c 54 1a
```
This is how u-boot handles it: [board/ti/common/board_detect.c](https://github.com/u-boot/u-boot/blob/master/board/ti/common/board_detect.c#L302)
```
	/* BeagleBone Green '1' eeprom, board_rev: 0x1a 0x00 0x00 0x00 */
	if (am_ep.version[0] == 0x1a && am_ep.version[1] == 0x00 &&
	    am_ep.version[2] == 0x00 && am_ep.version[3] == 0x00)
		strlcpy(ep->version, "BBG1", TI_EEPROM_HDR_REV_LEN + 1);
```

pdp7, 2020-01-22:
I have created pull request
add condition to detect BeagleBone Green #60

makermelissa, 2020-01-22:
Fixed via #60

closed 2020-01-22 by makermelissa