adafruit/Adafruit_Python_PlatformDetect #61: No Revision on cpuinfo

mruz, 2020-01-24:
On [openSUSE in Raspberry Pi 4](https://en.opensuse.org/HCL:Raspberry_Pi4) there is no `Revision` section in `cat /proc/cpuinfo`, so board isn't properly detected:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/0eb6058aa57cfd3d928eb724a3dcd36c66c61030/adafruit_platformdetect/board.py#L290-L294
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/0eb6058aa57cfd3d928eb724a3dcd36c66c61030/adafruit_platformdetect/board.py#L230-L232

https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/0eb6058aa57cfd3d928eb724a3dcd36c66c61030/adafruit_platformdetect/board.py#L119

As I see `cat /proc/device-tree/system/linux,revision | xxd` returns the rev that is present in the list https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/0eb6058aa57cfd3d928eb724a3dcd36c66c61030/adafruit_platformdetect/constants/boards.py#L295-L297

makermelissa, 2020-01-24:
This one is tricky because you're decoding a binary. I think the proper way of fixing this is to edit the `_pi_rev_code` function have it attempt to get the rev code the way it does now so that it doesn't break existing boards, check to see if it has a value and if not then attempt to decode the binary.

Also, it appears `cat /proc/device-tree/system/linux,revision | xxd` works on Raspbian as well, so no complicated testing setup.

makermelissa, 2020-01-24:
@mruz, if you would like to take a stab at fixing this and submitting a PR, I'd be happy to take a look. Otherwise, this may have to wait a bit.

commit 797bbbe8 (2020-02-11): Get revision from device-tree, fix #61
closed 2020-02-11 by makermelissa