adafruit/Adafruit_Python_PlatformDetect #66: advice for adding new board?

cman122887, 2020-03-01:
hello!

Thanks for making this, it's really cool.
I have a libre computer La Frite that runs on an Amlogic S805X.
When I ran the detector it looks like everything pretty much came out as None.
Digging through it seems like I will just have to make some files in the chips and boards files to add to this collection?
Please let me know if that is the way to go about it.
...
Just noticed some connected issues and saw that this info might be needed so I am putting this here in case:
`/proc/cpuinfo`

```
processor	: 0
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
```

`/proc/device-tree/model`
`Libre Computer Board AML-S805X-AC`

Thanks!

yeyeto2788, 2020-03-02:
Hey @cman122887,

I would first add your board and chip on the contants (https://github.com/adafruit/Adafruit_Python_PlatformDetect/tree/master/adafruit_platformdetect/constants)

Then, you just need to add the logic on the `board.py` and `chip.py` modules, take into account that we're already detecting a S805 chip (I guess your SBC has a different version of it) so you need to be careful so we can still detect previous boards.

The detection might be easier if for example you're running Armbian which sets a variable on the `/etc/armbian-release`.

Hope this helps you.

Regards.

cman122887, 2020-03-04:
Awesome, thanks for the info @yeyeto2788 
gonna try and send up a PR in a day or two, hopefully I don't mess this up.
Thanks

closed 2020-03-04 by cman122887