adafruit/Adafruit_Python_PlatformDetect #98: chip.py using very inefficient subcall to pyftdi

judule1, 2020-09-08:
Pardon any unfamiliarity, as I've only been poking around for a day. 

I'm currently using a Raspberry Pi 4B (up to date) with an Adafruit FT232H adapter (USB-C version) to an LIS3DH. While using the example Adafruit LIS3DH CircuitPython libraries, I discovered that I was getting a whopping 5 samples per second out of an expected 400 (default data rate). After a few hours of digging and tracing, I discovered that pyftdi is essentially crawling through and examining many USB devices every time I read bytes, leading to 120ms being spent _every single accelerometer read_ on just init backend calls through libusb1.py.

This is happening due to [line 72 in chip.py](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/366a41f603a8d22de1192c3fc480c870d5347fe9/adafruit_platformdetect/chip.py#L72): `count = len(UsbTools.find_all([(0x0403, 0x6014)]))`, which is there solely to determine whether the FT232H (or whatever board you might be using) is connected. The corresponding function in the [pyftdi library](https://github.com/eblot/pyftdi/blob/563930d9aa459f357de7d0e7492c73c692d44838/pyftdi/usbtools.py#L100) seems to do local caching which worked as I expected, so not sure why this kind of delay is happening in there - though there are a few TODO's in there about optimizing searches. Regardless, I didn't do much experimentation in there after I commented out the chip.py line as a hacky workaround (with the hope that I'll always be connected while running this & set the Blinka env variable in-application). My sampling rate went up to ~240 from 5 - still not where I'd like it to be, but much closer in magnitude.   

Anyhow, I'd like to make a newbie recommendation that the count() call be changed to something more efficient. Also, thanks for writing all these libraries!

(Note: everything installed via pip3)
Raspbian Buster Desktop (fully up to date as of Sep 7, 2020) 
pyftdi 0.52.2
Adafruit Blinka 5.3.3
adafruit-circuitpython-busdevice 5.0.1
Adafruit-PlatformDetect 2.15.0

ladyada, 2020-09-08:
ok gotta ask - why are you using an FTDI to talk to the LIS3DH instead of directly connecting to the Pi??

judule1, 2020-09-08:
Great question! Unfortunately I'm in a situation where my gateways are more easily deployed at some distance from my sensors (not just the LIS3DH). That length needs to be variable but will pretty much always be >1m, so SPI gets pretty dicey there.

ladyada, 2020-09-08:
@judule1 where's that chip id function called from? it should only be called once at the beginning of the script

judule1, 2020-09-08:
I'm away from my logs, but IIRC it's line 55 of board.py, also calling for id. It's called 14 times per acceleration read according to cProfile; not sure why that'd be.

ladyada, 2020-09-08:
@makermelissa ok wanna look? we should only check the chip id once

makermelissa, 2020-09-08:
Sure

PR #100 (2020-09-11): Implemented simple chip id and board id caching
Fixes #98. This adds a simple caching mechanism by storing the in the class and returning it from cache after the first time. I noted that it went from attempting to detect the FT232H 6 times down to 1 with this strategy when using `import board`. The root cause for the issue was that any time we called things like any_raspberry_pi_40_pin, it would have to fetch the chip ID, which was causing it to run all detection routines again. This avoids it unnecessarily running all the routines again, but gives it the value it needs.
closed 2020-09-11 by makermelissa