adafruit/Adafruit_Python_PlatformDetect #103: any_jetson_board() fails when running on "JETSON_NX"

andySigler, 2020-10-11:
The following test fails when running on Jetson Xavier NX:

```python
import adafruit_platformdetect

d = adafruit_platformdetect.Detector()
is_jetson = d.board.any_jetson_board

assert is_jetson is True
```

Seems likely it is caused by [this commit](https://github.com/adafruit/Adafruit_Python_PlatformDetect/commit/e3fede2f5c8ed0a2d3e78f6f3323ba87d5ac6d17).

ladyada, 2020-10-11:
@swarren hihi can you check it?

swarren, 2020-10-12:
https://github.com/swarren/Adafruit_Python_PlatformDetect/tree/fix-any_jetson_board probably fixes this. I'll test tomorrow.

swarren, 2020-10-12:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/106 should fix this; I've tested that it works with the code in this bug report.

ladyada, 2020-10-13:
thank you @swarren :)

closed 2020-10-13 by ladyada