adafruit/Adafruit_Python_PlatformDetect #112: Support for Allwinner H5

twa127, 2020-10-30:
Hi,

I'd like to add support for the Orange Pi Zero Plus2 H5, but am struggling to see how I can do it in chips.py as there is already detection,  the h5 returns sun50iw1p1 which is already being used for the A64

            elif "sun50iw1p1" in hardware:
                linux_id = chips.A64

Any idea how I can resolve the issue ?

makermelissa, 2020-10-30:
The A64 and H5 may be similar enough that you could just use that.

twa127, 2020-10-30:
Thanks for getting back to me, unfortunately the pin mapping is totally different, I have a local copy for the H5 based Orange Pi working fine, its just at I can see how to to have both the Pine A64 and the Orange Pi H5  at the same time, the problem being that both the A64 and H5 are identified as sun50iw1p1.

makermelissa, 2020-10-30:
Ok, it sounds like we'll have to narrow it down a bit further then. I just pulled out my Pine64. Let me see what I can find.

makermelissa, 2020-10-30:
Here's the results of `cat /proc/cpuinfo`
```
pi@pine64:~$ cat /proc/cpuinfo
processor	: 0
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
```

makermelissa, 2020-10-30:
Here's the results of `cat /proc/device-tree/compatible`
```
pine64,pine64-plusallwinner,sun50i-a64
```

makermelissa, 2020-10-30:
I'm thinking possibly one CPU fields differs or possibly the `sun50i-a64` inside of the device-tree/compatible results.

twa127, 2020-10-31:
cat/proc/cpuinfo

root@orangepizeroplus2-h5:/home/pi/pihome# cat /proc/cpuinfo
processor       : 0
Processor       : AArch64 Processor rev 4 (aarch64)
Hardware        : sun50iw1p1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
Processor       : AArch64 Processor rev 4 (aarch64)
Hardware        : sun50iw1p1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
Processor       : AArch64 Processor rev 4 (aarch64)
Hardware        : sun50iw1p1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
Processor       : AArch64 Processor rev 4 (aarch64)
Hardware        : sun50iw1p1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

twa127, 2020-10-31:
cat /proc/device-tree/compatible

xunlong,orangepi-zero-plus2allwinner,sun50i-h5

makermelissa, 2020-10-31:
Ok cool. I'm going to submit a PR that changes the A64 chip to use the dt-compatible value for detection and I would suggest you use that too.

PR #113 (2020-10-31): Change Pine64 to use dt-compatible
Fixes #112. Apparently the hardware value that was being checked has the same value for the H5 chip eventhough they're different. This fixes that.
closed 2020-10-31 by ladyada