adafruit/Adafruit_Python_PlatformDetect #127: Board not supported when detecting on Raspberry Pi Compute Module 4 rev 1.0

kouliss, 2021-01-01:
Hello,
when I run the script on Raspberry Pi Compute Module 4 rev 1.0, i am getting a "Board not supported" error message.

Here are the info from detect.py:
Chip id:  BCM2XXX
Board id:  None

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Black? False
Is this a BeagleBone Green? False
Is this a Giant Board? False
Is this a Coral Dev Board? False
Is this a Coral Dev Board Mini? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an STM32MP1 Board? False
Is this an OS environment variable special case? False
Raspberry Pi detected.

CPU info:
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : a03140
Serial          : 10000000a8b419ce
Model           : Raspberry Pi Compute Module 4 Rev 1.0

Is there any workaround for this error?

Thank You in advance for Your help.

Martin

makermelissa, 2021-01-04:
Hi thanks. I hadn't added the CM4 because I didn't have the hardware to retrieve numbers to add. Thanks for the info. That will be very helpful in getting this board added.

closed 2021-01-04 by makermelissa

timonsku, 2021-02-05:
Hi @kouliss could you provide me with the part number for this specific CM4 module? I'd be interest to see what hardware difference is dictating the different revision codes for the CM4.

kouliss, 2021-02-05:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Hello Timon,where can I get the part number? Is it printed somewhere on the CM4Martin                                                                                                                                                                           Od: notifications@github.comOdesláno: 5. února 2021 22:20Komu: Adafruit_Python_PlatformDetect@noreply.github.comOdpověď komu: reply@reply.github.comKopie: m.volek@excam.cz; mention@noreply.github.comPředmět: Re: [adafruit/Adafruit_Python_PlatformDetect] Board not supported when detecting on Raspberry Pi Compute Module 4 rev 1.0 (#127)  
Hi @kouliss could you provide me with the part number for this specific CM4 module? I'd be interest to see what hardware difference is dictating the different revision codes for the CM4.

makermelissa, 2021-02-05:
There's supposedly 32 variants, so I imagine some kind of pattern will soon emerge and we can get them all added.

timonsku, 2021-02-05:
@kouliss it is printed on the little paper packaging. If you don't habe that anymore but know the spec from the top of your head that would also help (wifi, ram, emmc)

kouliss, 2021-02-05:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Unfurtunetelly I don't have paper packing anymore. This is the  version with no wifi, 2gb ram and 16gb emmc.                                                                                                                                                                          Od: notifications@github.comOdesláno: 5. února 2021 22:39Komu: Adafruit_Python_PlatformDetect@noreply.github.comOdpověď komu: reply@reply.github.comKopie: m.volek@excam.cz; mention@noreply.github.comPředmět: Re: [adafruit/Adafruit_Python_PlatformDetect] Board not supported when detecting on Raspberry Pi Compute Module 4 rev 1.0 (#127)  
@kouliss it is printed on the little paper packaging. If you don't habe that anymore but know the spec from the top of your head that would also help (wifi, ram, emmc)

kouliss, 2021-02-05:
#response_container_BBPPID{font-family: initial; font-size:initial; color: initial;}  Is the package already updated? I tried updates and run the script, but it still reports "Board not supported"                                                                                                                                                                                                                                                                                                                Od: notifications@github.comOdesláno: 5. února 2021 22:36Komu: Adafruit_Python_PlatformDetect@noreply.github.comOdpověď komu: reply@reply.github.comKopie: m.volek@excam.cz; mention@noreply.github.comPředmět: Re: [adafruit/Adafruit_Python_PlatformDetect] Board not supported when detecting on Raspberry Pi Compute Module 4 rev 1.0 (#127)  
There's supposedly 32 variants, so I imagine some kind of pattern will soon emerge and we can get them all added.

PR #129 (2021-01-04): Added another Pi CM4 Rev ID
Fixes #127.