adafruit/Adafruit_Python_PlatformDetect #133: Pine64 board not being detected

makermelissa, 2021-01-18:
Hi thanks for adding the support to the Pine64! 

This doesn't work for me in my machine I see this for cat /proc/cpuinfo :
Processor	: AArch64 Processor rev 4 (aarch64)
processor	: 0
processor	: 1
processor	: 2
processor	: 3
Features	: fp asimd aes pmull sha1 sha2 crc32 
CPU implementer	: 0x41
CPU architecture: AArch64
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: Pine64+

and  cat /proc/device-tree/model 

Pine64+

Maybe is it the + at the end?

Thanks!

_Originally posted by @lucapinello in https://github.com/adafruit/Adafruit_Python_PlatformDetect/issues/52#issuecomment-761895525_

makermelissa, 2021-01-18:
See link in original issue. Basically they truncated the `/proc/device-tree/compatible` string on at least 1 build, so there's no way to accurately detect the processor.

lucapinello, 2021-01-18:
Thanks again Melissa. Btw which build are you currently using? I can quickly flash the SD card to install one that is compatible. There are so many options!

makermelissa, 2021-01-18:
If I recall correctly, I used Armbian.

makermelissa, 2021-01-18:
If you're not running Armbian, installing that will probably fix your issue. Also, just for the record, which distro are you running? I'd like this PR to reflect which distro it's not being detected in correctly.

lucapinello, 2021-01-18:
Hi Melissa, I just tried with a more recent version of Armbian (buster) and everything works fine now! I was using a very old distro, so feel free to close this. Thanks again!

closed 2021-01-18 by makermelissa