adafruit/Adafruit_Python_PlatformDetect #144: AttributeError: 'module' object has no attribute 'environ' on MicroPython and an ESP32

aziascreations, 2021-02-25:
Hello,

I was trying to replicate one of the examples from the "[adafruit/Adafruit_CircuitPython_SHTC3](https://github.com/adafruit/Adafruit_CircuitPython_SHTC3)" repository in the REPL, but when I imported the "busio" module I got an error from the [adafruit_platformdetect/chip.py](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/e0fe1b012898fa824944d6805ca74be0fa027968/adafruit_platformdetect/chip.py) file which returned this traceback:
```Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/busio.py", line 17, in <module>
  File "/lib/adafruit_blinka/agnostic/__init__.py", line 18, in <module>
  File "/lib/adafruit_platformdetect/chip.py", line 69, in id
AttributeError: 'module' object has no attribute 'environ'
```

<br>

It seems like it comes from the [line number 69](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/e0fe1b012898fa824944d6805ca74be0fa027968/adafruit_platformdetect/chip.py#L69) in that file where the script tries to access the `os.environ` attribute.

Which is also the case when I list them:
```
>>> import os
>>> dir(os)
['__class__', '__name__', 'remove', 'VfsFat', 'VfsLfs2', 'chdir', 'dupterm', 'dupterm_notify', 'getcwd', 'ilistdir', 'listdir', 'mkdir', 'mount', 'rename', 'rmdir', 'stat', 'statvfs', 'umount', 'uname', 'urandom']
```

<br>

I'm currently using the "ASDelivery ESP32 NodeMCU" with an ESP32-WROOM-32 on it which is running the prebuilt firmware from the official website named `esp32-idf4-20210202-v1.14.bin`. 

Is this normal or is it a bug, I haven't seen any mention of this error anywhere ?

ladyada, 2021-02-27:
yeah it probably isnt working on micropython due to pretty much nobody using it outside of linux. wanna try patching/fixing the library to see if it can be re-worked for micropython?

Oliver614, 2021-02-28:
I've also got the exact same issue. 
I'll try and have a look at it. 
In the meantime do you know if previous versions also have this issue?

ladyada, 2021-02-28:
its probably been broken for a long time :D

makermelissa, 2021-05-07:
Fixed via #161

closed 2021-05-07 by makermelissa