adafruit/Adafruit_Python_PlatformDetect #147: Board not supported None : EspressoBinBoard on OpenWrt

erdoukki, 2021-03-17:
Héllo,

How can I add or request an addition for the Marvell EspressoBin Board ?

Thanks...

```
# cat /proc/device-tree/mode
l 
Globalscale Marvell ESPRESSOBin Board V7
```
```
# cat /proc/device-tree/comp
atible 
globalscale,espressobin-v7globalscale,espressobinmarvell,armada3720marvell,armada3710
```
```
# cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 25.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 25.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

```

makermelissa, 2021-03-17:
You can follow this guide: https://learn.adafruit.com/adding-a-single-board-computer-to-platformdetect-for-blinka

erdoukki, 2021-03-20:
Thanks, I have added support for espressobin board to platform support but need now to add it also to the gpio support of another Adafruit library !
The cpu and board are detected, but sais as unsupported !
So complicated...

makermelissa, 2021-04-22:
Here's a link to the next guide: https://learn.adafruit.com/adding-a-single-board-computer-to-blinka

makermelissa, 2021-05-07:
Closing due to lack of response. Feel free to re-open or just submit a PR if you want to work on this. We don't have this board.

closed 2021-05-07 by makermelissa