adafruit/Adafruit_Python_PlatformDetect #150: Do not rely on specific distribution (armbian) to detect OrangePi and NanoPi platforms

razerraz, 2021-03-29:
Hi,

Running this on Nanopi or OrangePi without using armbian leads to board detection failure, as it leans on armbian release string from /etc/armbian-release file.
Seems to me that board detection can be done from various other available info, a least on a nanopi h3 and the orangepizero + (h5).
I can try to work on something and PR it when it become mature enough if you share the same goal not to be distro dependant for detection

Cheers

ladyada, 2021-03-29:
sure, if you can get it working on other distros without breaking armbian, we would review any PR

twa127, 2021-05-10:
Hi,

Does anyone have an objection if I fix pull request #151 as it would be useful for the Orange Pi Zero 2 running the Orange Pi Debian image ?

ladyada, 2021-05-10:
go for it :)

twa127, 2021-05-10:
Thanks, went for it :)

makermelissa, 2021-05-10:
Fixed via #164

closed 2021-05-10 by makermelissa