adafruit/Adafruit_Python_PlatformDetect #156: RASPBERRY_PI_ZERO_W not detected if board revision is 0000

4nd3r, 2021-05-01:
looks like kernel bug (upstream or debian), but maybe we want to work around it?

using image from https://raspi.debian.net/tested-images/ and tried with two different kernel:

<details>

```
# uname -r
4.19.0-14-rpi
# cat /proc/cpuinfo
processor       : 0 
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.34
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41 
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 0000
Serial          : 00000000d83bf83f

# uname -r
5.10.0-0.bpo.5-rpi
# cat /proc/cpuinfo 
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.34
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 0000
Serial          : 00000000d83bf83f
```

</details>

relevant code:
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/adafruit_platformdetect/board.py#L206

possible workaround:

```diff
--- board.py    2021-05-01 12:55:47.189499433 +0300
+++ /usr/local/lib/python3.7/dist-packages/adafruit_platformdetect/board.py     2021-05-01 12:59:37.766854833 +0300
@@ -205,7 +205,7 @@
             return None
         rev = self.detector.get_cpuinfo_field("Revision")
 
-        if rev is not None:
+        if rev is not None and not re.search('^0+$', rev):
             return rev
 
         try:
```

this way revision check will fail properly and proceeds with `/proc/device-tree/system/linux,revision`.

`detect.py` output without and with workaround:

```
# python3 detect.py
Chip id:  BCM2XXX
Board id:  None
--8<-- snip --8<--
Raspberry Pi detected.

# python3 detect.py
Chip id:  BCM2XXX
Board id:  RASPBERRY_PI_ZERO_W
--8<-- snip --8<--
Raspberry Pi detected.
```

but before I create PR, maybe maintainer(s) have some better ideas or additional thoughts?

ladyada, 2021-05-01:
yeah can you ask pi foundation why a board would have board revision 0000? do you see the same with raspbian? we really only support pios/raspbian

4nd3r, 2021-05-01:
ok, works with raspbian.

closed 2021-05-01 by 4nd3r

ladyada, 2021-05-01:
ideally whatever OS ya wanna use (debian) can add the revision code, its the only way we know to determine the exact board! :(