adafruit/Adafruit_Python_PlatformDetect #205: Add typing?

tekktrik, 2021-12-14:
I'm still going through libraries and adding typing, and just wanted to see if there was any benefit to adding those here.  I know this isn't necessarily a commonly end-user modified file, so wanted to check.

makermelissa, 2021-12-16:
Yeah, I think there is. It will help identify missing things and make it easier to add typing to supported libraries.

tekktrik, 2021-12-16:
Sounds good!  I'll get on it.

tekktrik, 2021-12-18:
I'm having some issues with cyclic imports and the best way to solve it I think would probably be to flatten `board.py` and `chip.py` into `__init__.py`.  Wanted to check in about doing so before actually submitting a PR with that change, since it's a pretty decent structural change.  Basically the problem is that I can't type the other two files' classes as taking `Detector` as an argument based on the current structure.

tekktrik, 2021-12-18:
Wanted to weigh that change against the benefit of typing

tekktrik, 2022-01-03:
Any thoughts on this?  I know it would be a somewhat large change, at least in terms of organization of code.  If it makes sense not to rearrange, it may warrant just closing this issue.

makermelissa, 2022-01-04:
The cyclic import issues may actually mean that the imports might need slight updating. Please don't flatten the files. That will break Blinka until it is updated there too and we recently went through a sizable bump when they were unflattened about a year ago and I'd prefer not to do that.

makermelissa, 2022-01-04:
If it's giving you trouble, I can go through and update... Just not today. I just redid all the Blinka Displayio stuff and that was way more complex than this. I think I also bumped into the cyclic import issue with that and just fixed it so it wasn't cyclically importing.

tekktrik, 2022-01-04:
Sounds good!

PR #219 (2022-02-21): Added Type Annotations
Fixes #205. Tested detect.py still runs properly on a raspberry pi 4.
closed 2022-02-21 by makermelissa