adafruit/Adafruit_Python_PlatformDetect #214: Pi4B , revision a03115 missing from Pi4B list,  not detected, causes raspi-blinka.py to report "None" and import board fail

joedellerfc, 2022-02-07:
I have circuitpython running on Pi Zero 2w, an original B+, but recently purchased a Pi4B 1G Model
and things aren't going so well, as the raspi-blinka.py script reports "None"detected" when it runs.
Although the script runs, I cannot import board without getting the NotImplemented error.
My Pi 4 id is a03115:

>>> import adafruit_platformdetect
>>> adafruit_platformdetect.Detector().get_cpuinfo_field("Revision")
'a03115'
>>> adafruit_platformdetect.Board(adafruit_platformdetect.Detector()).id
>>>
If I force it to be a Pi4:
pi@pi41:~ $ sudo BLINKA_FORCEBOARD=RASPBERRY_PI_4B python3
Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import board

All good.
Without the environment variable

>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/board.py", line 285, in <module>
    raise NotImplementedError(
NotImplementedError: Adafruit-PlatformDetect version 3.19.4 was unable to identify the board and/or microcontroller running the Linux platform. Please be sure you have the latest packages running: 'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

I have the following installed:
pi@pi41:~ $ sudo python3 -m pip list | grep Adafruit
Adafruit-ADS1x15                 1.0.2
Adafruit-Blinka                  6.20.1
Adafruit-PlatformDetect          3.19.4
Adafruit-PureIO                  1.1.9

I can see that my id of a03115 isn't in the Pi4 list, will do a local test to add it and see if that fixes the issue.

Regards,

Joe

ladyada, 2022-02-07:
ok please submit a PR to add the revision code when ya can :)

joedellerfc, 2022-02-07:
All done, thanks.

closed 2022-02-07 by joedellerfc