adafruit/Adafruit_Python_PlatformDetect #218: BeagleBone Black is not detected properly

Dario503, 2022-02-20:
debian@BeagleBone:~$ sudo python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from adafruit_platformdetect import Detector
>>> detector = Detector()
>>> print("Chip id: ", detector.chip.id)
Chip id:  AM33XX
>>> print("Board id: ", detector.board.id)
Board id:  None
>>> # Check for specific board models:
>>> print("Pi 3B+? ", detector.board.RASPBERRY_PI_3B_PLUS)
Pi 3B+?  False
>>> print("BBB? ", detector.board.BEAGLEBONE_BLACK)
BBB?  False
>>> print("Orange Pi PC? ", detector.board.ORANGE_PI_PC)
Orange Pi PC?  False
>>> print("generic Linux PC? ", detector.board.GENERIC_LINUX_PC)
generic Linux PC?  False
>>>

makermelissa, 2022-02-21:
Looks like the chip is right, but the board isn't.

RobertCNelson, 2022-02-22:
Hi @makermelissa , i was also talking with @pdp7 about this... Maybe with the BeagleBone family we are trying to do too much... We try to support v3.8.x  -> v5.17.x... @Dario503 found another interesting mainline change, as our eeprom location changed on mainline. ;)

```
sys/bus/nvmem/devices/0-00500/nvmem -> sys/bus/nvmem/devices/0-00501/nvmem
```

Any thoughts to allow our users to just hardcode it's a BeagleBone, and then define which interfaces,  spi/i2c/etc, they want to use?

```
root@BeagleBone:~# uname -r
5.15.23-bone20
root@BeagleBone:~# tree /sys/bus/nvmem/devices/
/sys/bus/nvmem/devices/
├── 0-00501 -> ../../../devices/platform/ocp/44c00000.interconnect/44c00000.interconnect:segment@200000/44e0b000.target-module/44e0b000.i2c/i2c-0/0-0050/0-00501
└── omap_rtc_scratch0 -> ../../../devices/platform/ocp/44c00000.interconnect/44c00000.interconnect:segment@200000/44e3e074.target-module/44e3e000.rtc/omap_rtc_scratch0

2 directories, 0 files
root@BeagleBone:~# xxd /sys/bus/nvmem/devices/0-00501/nvmem
00000000: aa55 33ee 4133 3335 424e 4c54 4747 3141  .U3.A335BNLTGG1A
00000010: 3230 3230 3036 3030 3033 3537 ffff ffff  202006000357....
00000020: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00000030: ffff ffff ffff ffff ffff ffff 3243 4637  ............2CF7
00000040: 4631 3036 3030 3031 0000 0000 0000 ffff  F1060001........

"0-00500"... is also not stable!!
```

The one solution, i don't want to do... I say only "v5.10.x" is supported, and then once a year we bump things..

The other crazy idea is epcohs, 3.8.x/bbb -> x, 4.14.x/bbb -> y, 5.4.x/bbb ->z ... but would be a mess...

Regards,

ladyada, 2022-02-22:
hi there is *no* way to determine when code is running on a beaglebone?

makermelissa, 2022-02-22:
I'm spinning up a BBB at the moment. Maybe we could do something like I did for Beaglebone AI: https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/main/adafruit_platformdetect/board.py#L257-L262

makermelissa, 2022-02-22:
Ok, looks like running detect.py works fine:
```
debian@beaglebone:~/Adafruit_Python_PlatformDetect/bin$ python3 detect.py
Chip id:  AM33XX
Board id:  BEAGLEBONE_BLACK

Is this a DragonBoard 410c? False
Is this a Pi 3B+? False
Is this a Pi 4B? False
Is this a 40-pin Raspberry Pi? False
Is this a Raspberry Pi Compute Module? False
Is this a BeagleBone Board? True
Is this a Giant Board? False
Is this a Coral Dev Board? False
Is this a Coral Dev Board Mini? False
Is this a MaaXBoard? False
Is this a MaaXBoard Mini? False
Is this a SiFive Unleashed?  False
Is this a PYNQ Board? False
Is this a Rock Pi board? False
Is this a NanoPi board? False
Is this a Clockwork Pi board? False
Is this an embedded Linux system? True
Is this a generic Linux PC? False
Is this a UDOO Bolt? False
Is this an ASUS Tinker Board? False
Is this an STM32MP1 Board? False
Is this an OS environment variable special case? False
```
I'm, using the latest. Perhaps you're right about trying to support too wide of a range.

makermelissa, 2022-02-22:
Oh, I guess it's not the latest. Just the latest recommended image with 4.19.94 kernel.

RobertCNelson, 2022-02-22:
Give me 20 minutes, i'll have a kernel version vs location off eeprom database setup.. ;)

Just need to push an update and see what we have..
http://gfnd.rcn-ee.org:81/farm/uptime/

Regards,

makermelissa, 2022-02-22:
So yeah, looks like peeking at `/proc/device-tree-model` may be a workable alternative:
```
debian@beaglebone:~/Adafruit_Python_PlatformDetect/bin$ cat /proc/device-tree/model
TI AM335x BeagleBone Black
```

RobertCNelson, 2022-02-22:
Personally... i've started moving away from that... too many clones...

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L136-L232

makermelissa, 2022-02-22:
Ok, if you'd like to modify https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/main/adafruit_platformdetect/board.py#L219 to also check at the new location (probably right after checking the first location) and submit a PR, we'd be happy to take a look.

RobertCNelson, 2022-02-22:
Okay, @makermelissa this is what i'm seeing:

```
tree /sys/bus/nvmem/devices/

4.19.94-ti-r71
4.19.94-ti-rt-r71
5.4.106-ti-r38
5.4.106-ti-rt-r38
5.4.161-bone63
5.4.161-bone-rt-r63

0-00500

tree /sys/bus/nvmem/devices/

5.10.90-ti-r36
5.10.90-ti-rt-r36
5.10.90-bone60
5.10.90-bone-rt-r60
5.15.23-bone20
5.15.23-bone20

0-00501

```

So something in teh kerel ABI change... 0-00500 -> 0-00501  !!!

makermelissa, 2022-02-22:
Good to know. We should have it just try the location it does now. If that fails, it should just try the second location. That way it just works for the different kernels.

RobertCNelson, 2022-02-22:
i agree, that should be the safest for now.. i was little worried some where even more off..

makermelissa, 2022-03-01:
It looks like you need elevated permissions to read the new location:
```
debian@beaglebone:~$ cat /sys/bus/nvmem/devices/0-00501/nvmem
cat: /sys/bus/nvmem/devices/0-00501/nvmem: Permission denied
```
Python isn't wanting to read it either without prepending a sudo.

RobertCNelson, 2022-03-01:
Hi @makermelissa , I wonder if just fixed this, what is the permission of the directory..

ls -lh of that dir....

makermelissa, 2022-03-01:
```
debian@beaglebone:~$ ls -lh /sys/bus/nvmem/devices/0-00501
lrwxrwxrwx 1 root root 0 Jan  1  2000 /sys/bus/nvmem/devices/0-00501 -> ../../../devices/platform/ocp/44c00000.interconnect/44c00000.interconnect:segment@200000/44e0b000.target-module/44e0b000.i2c/i2c-0/0-0050/0-00501
debian@beaglebone:~$ ls -lh /sys/bus/nvmem/devices/0-00501/nvmem
-rw------- 1 root root 0 Mar  1 19:22 /sys/bus/nvmem/devices/0-00501/nvmem
```

makermelissa, 2022-03-01:
Or maybe you meant this:
```
debian@beaglebone:/sys/bus/nvmem/devices/0-00501$ ls -lh
total 0
-rw------- 1 root root    0 Mar  1 19:22 nvmem
lrwxrwxrwx 1 root root    0 Mar  1 19:24 of_node -> ../../../../../../../../../../firmware/devicetree/base/ocp/interconnect@44c00000/segment@200000/target-module@b000/i2c@0/baseboard_eeprom@50
drwxr-xr-x 2 root root    0 Mar  1 19:24 power
lrwxrwxrwx 1 root root    0 Jan  1  2000 subsystem -> ../../../../../../../../../../bus/nvmem
-r--r--r-- 1 root root 4.0K Mar  1 19:24 type
-rw-r--r-- 1 root root 4.0K Jan  1  2000 uevent
```

RobertCNelson, 2022-03-01:
@makermelissa what version of bb-customizations does that image have?

```
root@BeagleBone:~# dpkg --list | grep bb-customizations
ii  bb-customizations                          1.20220119.0-0~bullseye+20220125              armhf        beagleboard.org customizations
```

```
root@BeagleBone:~# ls -lah /sys/bus/nvmem/devices/0-00501/
total 0
drwxrwxr-x 3 root gpio    0 Jan  1  2000 .
drwxrwxr-x 4 root gpio    0 Jan  1  2000 ..
-rw-rw---- 1 root gpio  32K Jan  1  2000 nvmem
lrwxrwxrwx 1 root gpio    0 Mar  1 03:08 of_node -> ../../../../../../../../../../firmware/devicetree/base/ocp/interconnect@44c00000/segment@200000/target-module@b000/i2c@0/baseboard_eeprom@50
drwxrwxr-x 2 root gpio    0 Jan  1  2000 power
lrwxrwxrwx 1 root gpio    0 Mar  1 03:08 subsystem -> ../../../../../../../../../../bus/nvmem
-r--r--r-- 1 root gpio 4.0K Jan  1  2000 type
-rw-rw-r-- 1 root gpio 4.0K Jan  1  2000 uevent
```

okay.... i'm not sure now where the udev rule is triggered..

```
root@BeagleBone:~# cat /etc/udev/rules.d/* | grep eeprom
```

REgards,

RobertCNelson, 2022-03-01:
```
debian@BeagleBone:~$ ls -lah /sys/bus/nvmem/devices/0-00501/
total 0
drwxrwxr-x 3 root gpio    0 Jan  1  2000 .
drwxrwxr-x 4 root gpio    0 Jan  1  2000 ..
-rw-rw---- 1 root gpio  32K Jan  1  2000 nvmem
lrwxrwxrwx 1 root gpio    0 Mar  1 03:08 of_node -> ../../../../../../../../../../firmware/devicetree/base/ocp/interconnect@44c00000/segment@200000/target-module@b000/i2c@0/baseboard_eeprom@50
drwxrwxr-x 2 root gpio    0 Jan  1  2000 power
lrwxrwxrwx 1 root gpio    0 Mar  1 03:08 subsystem -> ../../../../../../../../../../bus/nvmem
-r--r--r-- 1 root gpio 4.0K Jan  1  2000 type
-rw-rw-r-- 1 root gpio 4.0K Jan  1  2000 uevent
```

```
debian@BeagleBone:~$ cat /sys/bus/nvmem/devices/0-00501/nvmem | xxd
00000000: aa55 33ee 4133 3335 424e 4c54 3030 4330  .U3.A335BNLT00C0
00000010: 3235 3136 4242 424b 3236 3236 ffff ffff  2516BBBK2626....
00000020: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00000030: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00000040: ffff ffff ffff ffff ffff ffff ffff ffff  ................
00000050: 3430 3035 3234 2b31 3330 312b 3030 3937  400524+1301+0097
00000060: 3636 2b31 3632 3558 5858 5858 5858 5858  66+1625XXXXXXXXX
```

```
debian@BeagleBone:~$ cat /etc/dogtag 
BeagleBoard.org Debian Bullseye IoT Image 2022-02-28
```

makermelissa, 2022-03-01:
Ok, that looks promising. 
```
debian@beaglebone:/sys/bus/nvmem/devices/0-00501$ cat /etc/udev/rules.d/* | grep eeprom
# From: https://github.com/s-light/pocketbeagle_python_tests/blob/master/80-eeprom-noroot.rules
# /etc/udev/rules.d/80-eeprom-noroot.rules
# allow i2c group to read of eeprom
# /sys/bus/i2c/devices/0-0050/eeprom
# ls -l /sys/bus/i2c/devices/0-0050/eeprom /sys/bus/nvmem/devices/0-00500/nvmem
SUBSYSTEM=="i2c", DEVPATH=="*0-0050", RUN+="/bin/chown root:gpio /sys$env{DEVPATH}/eeprom", RUN+="/bin/chmod 0640 /sys$env{DEVPATH}/eeprom"
```

After a quick edit to the udev rules and a reboot it works.

RobertCNelson, 2022-03-01:
building right now... https://github.com/rcn-ee/repos/commit/f78bcd15939bf7486b91a8a3677cc35842c72938

makermelissa, 2022-03-01:
I'll submit a PR. Looks like it working great now. Users who upgrade will need to edit udev ruls though. File is `/etc/udev/rules.d/80-eeprom-noroot.rules` and just needs to be changed from **0-00500** to **0-00501**
```
debian@beaglebone:~/Adafruit_Python_PlatformDetect/bin$ python3 detect.py
Chip id:  AM33XX
Board id:  BEAGLEBONE_BLACK

```

makermelissa, 2022-03-01:
> building right now... [rcn-ee/repos@f78bcd1](https://github.com/rcn-ee/repos/commit/f78bcd15939bf7486b91a8a3677cc35842c72938)

even better

RobertCNelson, 2022-03-01:
apt repo is now updated...

Should be fixed with:

```
sudo apt update
sudo apt upgrade
```

i was going to tag and update:

https://forum.beagleboard.org/t/debian-10-x-buster-monthly-snapshots/31203

today, but i'll delay it so this fix will our buster image..

Regards,

closed 2022-03-01 by makermelissa

bgigous, 2024-12-11:
I'm still having an issue with a BeagleBone Green on kernel 5.10. I was able to get PlatformDetect to work when I tried `import board` previously with the changes [here](https://github.com/adafruit/Adafruit_Blinka/issues/886), however it seems that somehow the nvmem device changed from 0-00501 to 0-00502:

```bash
debian@BeagleBone:~$ tree /sys/bus/nvmem/devices/
/sys/bus/nvmem/devices/
|-- 0-00502 -> ../../../devices/platform/ocp/44c00000.interconnect/44c00000.interconnect:segment@200000/44e0b000.target-module/44e0b000.i2c/i2c-0/0-0050/0-00502
|-- 2-00540 -> ../../../devices/platform/ocp/48000000.interconnect/48000000.interconnect:segment@100000/4819c000.target-module/4819c000.i2c/i2c-2/2-0054/2-00540
`-- omap_rtc_scratch0 -> ../../../devices/platform/ocp/44c00000.interconnect/44c00000.interconnect:segment@200000/44e3e074.target-module/44e3e000.rtc/omap_rtc_scratch0

4 directories, 0 files
```

How would this have happened? I have not made any kernel changes or u-boot configuration changes since I added the udev rule linked above. So I'm not sure what has caused this change, and as a result, my board cannot be detected as a BBG like I would expect.

```
debian@BeagleBone:~$ sudo cat /sys/bus/nvmem/devices/0-00502/nvmem | xxd | head -n 2
00000000: aa55 33ee 4133 3335 424e 4c54 4242 4731  .U3.A335BNLTBBG1
00000010: 4242 4731 3233 3235 3033 3034 ffff ffff  BBG123250304....
```

beagle-version:
```
eeprom:[A335BNLTBBG1BBG123250304]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Bookworm IoT Image 2023-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2022.04-00038-gbaca7b469d (Nov 06 2024 - 17:14:19 -0700)]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0.kernel]
UBOOT: Loaded Overlay:[BB-I2C2-RTC-DS3231]
UBOOT: Loaded Overlay:[BB-SPIDEV0-00A0.kernel]
UBOOT: Loaded Overlay:[BB-UART1-00A0.kernel]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0.kernel]
kernel:[5.10.168-ti-r77]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=BB-I2C2-RTC-DS3231.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=BB-UART1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr6=BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
WARNING:pkg:[bb-cape-overlays]:[NOT_INSTALLED]
pkg:[bb-customizations]:[1.20240119.0-0~bookworm+20240119]
pkg:[bb-usb-gadgets]:[1.20231003.1-0~bookworm+20231003]
pkg:[bb-wl18xx-firmware]:[1.20230414.0-0~bookworm+20230414]
pkg:[kmod]:[30+20221128-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20230210-5]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input render netdev i2c bluetooth gpio admin tisdk weston-launch cloud9ide spi]
cmdline:[console=ttyS0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   13.473831] remoteproc remoteproc0: wkup_m3 is available
[   26.608058] systemd[1]: Reached target remote-fs.target - Remote File Systems.
[   53.394198] remoteproc remoteproc0: powering up wkup_m3
[   53.394236] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[   53.394534] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   81.494612] remoteproc remoteproc1: 4a334000.pru is available
[   81.556819] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   81.494612] remoteproc remoteproc1: 4a334000.pru is available
[   81.556819] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pinctrl-single
[   12.847925] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[   12.849823] gpio-of-helper ocp:cape-universal: Failed to get gpio property of 'P8_03'
[   12.849852] gpio-of-helper ocp:cape-universal: Failed to create gpio entry
[   13.729509] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0 name='P8_03'
[... 49 lines cut ...]
[   13.740550] gpio-of-helper ocp:cape-universal: Allocated GPIO id=50 name='P9_17'
[   13.740757] gpio-of-helper ocp:cape-universal: Allocated GPIO id=51 name='P9_18'
[   13.740945] gpio-of-helper ocp:cape-universal: Allocated GPIO id=52 name='P9_19'
[   13.741144] gpio-of-helper ocp:cape-universal: Allocated GPIO id=53 name='P9_20'
[   13.741333] gpio-of-helper ocp:cape-universal: Allocated GPIO id=54 name='P9_21'
[   13.741541] gpio-of-helper ocp:cape-universal: Allocated GPIO id=55 name='P9_22'
[   13.741757] gpio-of-helper ocp:cape-universal: Allocated GPIO id=56 name='P9_23'
[   13.741969] gpio-of-helper ocp:cape-universal: Allocated GPIO id=57 name='P9_24'
[   13.742500] gpio-of-helper ocp:cape-universal: Allocated GPIO id=58 name='P9_25'
[   13.742740] gpio-of-helper ocp:cape-universal: Allocated GPIO id=59 name='P9_26'
[   13.742954] gpio-of-helper ocp:cape-universal: Allocated GPIO id=60 name='P9_27'
[   13.743162] gpio-of-helper ocp:cape-universal: Allocated GPIO id=61 name='P9_28'
[   13.743356] gpio-of-helper ocp:cape-universal: Allocated GPIO id=62 name='P9_29'
[... cut ...]
[   13.743879] gpio-of-helper ocp:cape-universal: Allocated GPIO id=64 name='P9_31'
[   13.744115] gpio-of-helper ocp:cape-universal: Allocated GPIO id=65 name='P9_41'
[   13.744321] gpio-of-helper ocp:cape-universal: Allocated GPIO id=66 name='P9_91'
[   13.744555] gpio-of-helper ocp:cape-universal: Allocated GPIO id=67 name='P9_42'
[   13.744768] gpio-of-helper ocp:cape-universal: Allocated GPIO id=68 name='P9_92'
[   13.744782] gpio-of-helper ocp:cape-universal: ready
dmesg | grep wlcore
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

PR #221 (2022-03-01): Fixed Beaglebone Black detection for Kernel 5.10 or higher
Fixes #218.