adafruit/Adafruit_Python_PlatformDetect #225: SyntaxError: future feature annotations is not defined

H-Y-H-Y-H, 2022-03-06:
I was using this package and it worked perfectly. Now I have some trouble because I re-install the latest version which has "from __future__ import annotations" in both board.py and chip.py files. 

Below shows the error I got:

    Traceback (most recent call last):
      File "servo_motor.py", line 1, in <module>
        from adafruit_servokit import ServoKit
      File "/home/nvidia/Desktop/face_env/lib/python3.6/site-packages/adafruit_servokit.py", line 35, in <module>
        import board
      File "/home/nvidia/Desktop/face_env/lib/python3.6/site-packages/board.py", line 39, in <module>
        import adafruit_platformdetect.constants.boards as ap_board
      File "/home/nvidia/Desktop/face_env/lib/python3.6/site-packages/adafruit_platformdetect/__init__.py", line 11, in <module>
        from .chip import Chip
      File "/home/nvidia/Desktop/face_env/lib/python3.6/site-packages/adafruit_platformdetect/chip.py", line 23
        from __future__ import annotations
        ^
    SyntaxError: future feature annotations is not defined

I solved this by copying the previous version code.

dhalbert, 2022-03-06:
adafruit-blinka is now requiring Python 3.7. This library should also be requiring 3.7.

Which board and OS are you using that only has Python 3.6? Are you able to upgrade?

closed 2022-03-07 by dhalbert

dolikemedo, 2022-11-19:
@yulu54  Hi, How did you solve this? I'm trying to install older versions of adafruit-circuitpython-ads1x15 and it has the same error. I don't understand how older version has the same error. I tried installing even the oldest versions with pip3 and still the same errors. I have python3.6.9 in a container and It caused other issues upgrading python. Thanks for your answer!

zyth0x-1, 2022-12-16:
I got past some of these issues with adafruit-io by forcing a lower version of setuptools-scm with `pip install setuptools-scm==6.0.1 ` hope this helps someone

PR #226 (2022-03-07): Update Required Python to 3.7
Fixes #225.