adafruit/Adafruit_Python_PlatformDetect #229: About Error SyntaxError: future feature annotations.... on python 3.6

ctm6100, 2022-03-28:
### about the Error
I am a Jetson user (using pyhton 3.6), for Adafruit_Python_PlatformDetect version bigger than 3.19.6
the following error wil be recived
```python
>>> import board
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/board.py", line 39, in <module>
    import adafruit_platformdetect.constants.boards as ap_board
  File "/usr/local/lib/python3.6/dist-packages/adafruit_platformdetect/__init__.py", line 10, in <module>
    from .board import Board
  File "/usr/local/lib/python3.6/dist-packages/adafruit_platformdetect/board.py", line 24
    from __future__ import annotations
    ^
SyntaxError: future feature annotations is not defined
```
### Issues
Maybe this due to the [3.20.0](https://github.com/adafruit/Adafruit_Python_PlatformDetect/tree/3.20.0) added Type Annotations?
Therefore maybe the version that require Python 3.7 is starting from `3.20.0` instead of `3.21.0`?

### Possible quick fix for this issues
the solution that work for me (using python3.6)

1. remove the current version(anything > 3.19.6)
`$sudo -H pip3 uninstall Adafruit-PlatformDetect`  
2. specify the version to 3.19.6 
`$sudo -H pip3 install Adafruit-PlatformDetect==3.19.6`

OR 
try [backport of __future__ annotations](https://pypi.org/project/future-annotations/)(not yet test)
OR
update to python3.7(painful in jetson)(not yet test)

dhalbert, 2022-03-28:
Thanks, it sounds like maybe we should make a 3.20.1 release that requires 3.7, and withdraw the 3.20.0 release so it isn't found.

Is it possible in Jetson to install 3.7 or higher for user-only use, without replacing /usr/bin/python? Is there a package available or would you need to build it yourself?

ctm6100, 2022-03-28:
yes it is possible. In fact, it is petty easy 
```bash 
sudo apt-get install -y python3.7
# update the python3 priority
sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.6 1
sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.7 2
# view the change
sudo update-alternatives --config python
```
However, the painful part is to re-config or re-install or build from source  those package such as openCV, ROS, TensorRT, Pytorch, etc. on 3.7.

makermelissa, 2022-03-28:
PR #230 may fix this.

dhalbert, 2022-03-29:
If there are annotations in the imported CircuitPython code, then I think it the imports will fail, unfortunately, just further on.

Maybe the solution is on the Jetson to add this file in a PYTHONPATH-accessible place?
__future__.py:
```py
annotations = False
```
This will not work if there are other uses of `from __future__ import` in other Python code that you are using.

makermelissa, 2022-03-30:
I have updated the Jetson Nano Guide for installing Python 3.7 and making it the default, which actually was pretty easy:
https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup#install-python-3-dot-7-and-make-default-3114929-19

yulu54, 2022-05-06:
> I have updated the Jetson Nano Guide for installing Python 3.7 and making it the default, which actually was pretty easy: https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup#install-python-3-dot-7-and-make-default-3114929-19
So, Is that any problem at Opencv, numpy old version. Or are you update Opencv for python 3.7

makermelissa, 2022-05-06:
Does Opencv not run on Python 3.7?

yulu54, 2022-05-06:
> Does

I'll try tomorrow 👍

yulu54, 2022-05-07:
> ### about the Error
> I am a Jetson user (using pyhton 3.6), for Adafruit_Python_PlatformDetect version bigger than 3.19.6 the following error wil be recived
>
> ```python
> Traceback (most recent call [...]

Thank you, downgrade is solve the problem

makermelissa, 2022-05-07:
> Thanks, it sounds like maybe we should make a 3.20.1 release that requires 3.7.
I'm not sure how that can be done at this point in time. Perhaps your git-fu is better than mine. I know we could remove both the 3.20.0 and 3.20.1 releases though, since they are the only 2 in the gap where it's broken.

tekktrik, 2022-09-07:
What's the status of this?  Blinka is now 3.7+, right?

makermelissa, 2022-09-07:
> What's the status of this? Blinka is now 3.7+, right?

Correct

makermelissa, 2022-09-07:
> Thank you, downgrade is solve the problem

Closing for now since the problem was solved.

closed 2022-09-07 by makermelissa