adafruit/Adafruit_Python_PlatformDetect #238: Really old Beaglebone black

rbalsan1, 2022-08-17:
I found a stack of 30 BBBs circa 2014 at my job
eeprom: A335BNLT0A6A5113BBBK0690
when I run some sample commands
>>> from adafruit_platformdetect import Detector
>>> detector = Detector()
>>> print("Chip id: ", detector.chip.id)
Chip id:  AM33XX
>>> print("Board id: ", detector.board.id)
Board id:  None

I assume the eeprom code isn't in the list

ladyada, 2022-08-17:
yep - you could try looking thru the code around here
https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/main/adafruit_platformdetect/board.py#L226
see if you can figure out where to add support! a PR would be great - also make sure you're running the latest OS :)

rbalsan1, 2022-08-18:
I made a pull request that added the board to the list of boards seems to resolve the issue

closed 2022-08-18 by rbalsan1