adafruit/Adafruit_Python_PlatformDetect #242: Adafruit Blinka fails to detect Orange Pi Zero 2 correctly

laserwelder, 2022-08-29:
adafruit_platformdetect detects the AllWinner H616 chip correclly.  But detector.board.id reports "none".
Orange Pi Zero 2 is already defined in the constants file boards.py.
But the board.py module has some complex logic that should detect an Orange Pi Zero 2 but seems to fail.

There is only one SBC with an AllWinner H616 chip that I could find - so rather than fixing the logic a simple change to board.py setting the board to "ORANGE_PI_ZERO_2" whenever the chip is identified as H616, and removing all other code referencing H616 and ORANGE_PI_ZERO_2 from board.py seems to fix the problem.

laserwelder, 2022-08-29:
> Hi, I moved the issue to the appropriate repository where the detection occurs. After looking at the code, this sounds like a reasonable fix for the issue. We'd be happy to take a look if you'd like to submit a Pull Request.

Thanks for the fast reply Melissa. I'm sorry for posting in the wrong place, this was my first time posting on GitHub so any feedback on correct procedures is welcome. I will look at posting a pull request as you suggest, but first I will check if I have contributed to the issue. I am running Debian not armbian, for some reason the orangepi manufacturer recommends strongly against armbian. But today I am going to install armbian and see if the issue goes away. It it does then the simplification I suggested in platform detect might still be worth it, but is not as urgent.

laserwelder, 2022-08-30:
I can confirm that under Armbian this issue disappears.  Chip and Board on an Orange Pi Zero 2 are identified correctly.  Testing took a while because the current Armbian versions for the Orange Pi Zero 2 won’t boot.  This is a known issue and can be solved by installing an older version.  I was able to boot using Armbian_22.02.2_Orangepizero2_focal_legacy_4.9.255.

My apologies for raising an issue that was caused by my using a different distro than the one recommended.

laserwelder, 2022-08-30:
I am going to mark the issue closed.

closed 2022-08-30 by laserwelder

makermelissa, 2022-08-30:
No worries. I'm glad you were able to figure out the cause of issue.