adafruit/Adafruit_Python_PlatformDetect #243: Pi4B not recognized

craigerl, 2022-09-06:
I have a user using a raspberry pi image which is known to work in pi3b, rpizero, rpizero2w, however on his Pi4B, the board is not identified, and there is an unknown attribute towards the middle of the big if/elsif statement,

pi@digipi:~ $ ./direwatch.py -1 /run/direwatch.log

Traceback (most recent call last):

  File "/home/pi/./direwatch.py", line 44, in <module>

    import board

  File "/usr/local/lib/python3.9/dist-packages/board.py", line 121, in <module>

    elif board_id == ap_board.ORANGE_PI_3:

AttributeError: module 'adafruit_platformdetect.constants.boards' has no attribute 'ORANGE_PI_3'

makermelissa, 2022-09-06:
Make sure you have the latest platformdetect installed by running `pip3 install --upgrade Adafruit-PlatformDetect`

closed 2022-09-12 by makermelissa

craigerl, 2022-09-12:
Yes, confirmed, the update fixed the pi4b recognition - thank you!