adafruit/Adafruit_Python_PlatformDetect #273: Unable to install but it was working previously

bsteiner-dandy, 2023-01-18:
I'm getting an error trying to install on a Coral Development Board, but I know this worked previously because I had the install working in my setup script and running on multiple boards. Did something change in the last month or so that would affect this? The error seems to be due to `Invalid version: '7.5.3-deb10u1'`. Any ideas?

Full output:
```
mendel@robot-9:~$ pip3 install --upgrade setuptools
Requirement already satisfied: setuptools in /usr/local/lib/python3.7/dist-packages (66.0.0)
mendel@robot-9:~$ pip3 install --upgrade pip
Requirement already satisfied: pip in ./.local/lib/python3.7/site-packages (22.3.1)
mendel@robot-9:~$ pip3 install Adafruit-PlatformDetect
Collecting Adafruit-PlatformDetect
  Using cached Adafruit-PlatformDetect-3.39.0.tar.gz (42 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [28 lines of output]
      Traceback (most recent call last):
        File "<string>", line 36, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-id1y60fo/adafruit-platformdetect_ffba78b3db054114835210653d2ed33d/setup.py", line 49, in <module>
          packages=["adafruit_platformdetect", "adafruit_platformdetect.constants"],
        File "/usr/local/lib/python3.7/dist-packages/setuptools/__init__.py", line 86, in setup
          _install_setup_requires(attrs)
        File "/usr/local/lib/python3.7/dist-packages/setuptools/__init__.py", line 80, in _install_setup_requires
          dist.fetch_build_eggs(dist.setup_requires)
        File "/usr/local/lib/python3.7/dist-packages/setuptools/dist.py", line 877, in fetch_build_eggs
          replace_conflicting=True,
        File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 816, in resolve
          req, best, replace_conflicting, env, installer, required_by, to_activate
        File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 844, in _resolve_dist
          env = Environment(self.entries)
        File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 1044, in __init__
          self.scan(search_path)
        File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 1077, in scan
          self.add(dist)
        File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 1096, in add
          dists.sort(key=operator.attrgetter('hashcmp'), reverse=True)
        File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 2631, in hashcmp
          self.parsed_version,
        File "/usr/local/lib/python3.7/dist-packages/pkg_resources/__init__.py", line 2678, in parsed_version
          self._parsed_version = parse_version(self.version)
        File "/usr/local/lib/python3.7/dist-packages/pkg_resources/_vendor/packaging/version.py", line 266, in __init__
          raise InvalidVersion(f"Invalid version: '{version}'")
      pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: '7.5.3-deb10u1'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

tekktrik, 2023-01-19:
I can't figure out where `7.5.3-deb10u1` is coming from, it shouldn't be this package since we're using `setuptools_scm` and we haven't that version number for this package... Does it resolve if you use `--no-cache-dir`?

bsteiner-dandy, 2023-01-19:
It looks like it came from the package `reportbug`. I was able to create a workaround by uninstalling this package prior to installing the adafruit libraries. But I am not sure if I actually need the `reportbug` package or not, and I am unable to reinstall it after I uninstall it

tekktrik, 2023-01-19:
Weird, using a Debian package manager you should be able to install a version that perhaps follows [PEP 440](https://peps.python.org/pep-0440/) like `1.2.3`, though I didn't think it would check a Debian package's version.  I believe the current stable (`[7.10.3+deb11u1](https://packages.debian.org/source/stable/reportbug)) is complaint with the PEP, for what it's worth, in case you find more issues like this (minus signs are bad, but plus signs are okay).

bsteiner-dandy, 2023-01-19:
Ok after that PR went in it looks like its working now without uninstalling `reportbug` first. Thanks!

bsteiner-dandy, 2023-01-19:
Nevermind it looks like I still need to remove `reportbug` for this to work

tekktrik, 2023-01-19:
That PR hasn't been propagated to PyPI yet, sorry!  I'm honestly not sure that it will fix it though, to be honest.  I would try upgrade `reportbug` to that version I mentioned, if possible.

bsteiner-dandy, 2023-01-19:
I don't think I can upgrade it since I am stuck with Debian 10 and I think that is the Debian 11 version

tekktrik, 2023-01-19:
Actually, #274 may solve this because it also adds built Python wheels for download from PyPI which means it doesn't need to install locally anymore.

bsteiner-dandy, 2023-01-19:
Sound good, let me know when it is available via PyPI?

tekktrik, 2023-02-03:
@bsteiner-dandy it's now available on PyPI, does it fix your issue?  It's a pure Python wheel.

bsteiner-dandy, 2023-02-13:
looks like it fixed it!

closed 2023-02-13 by bsteiner-dandy