adafruit/Adafruit_Python_PlatformDetect #276: Build fails

makermelissa, 2023-01-30:
See https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/4047615129/jobs/6961847839. It might be an issue of running Python 3.11 in GitHub Actions.

Output:

```
Run find -type f -not -path "./.*" -not -path "./docs*" \( -name "*.py" -o -name "*.toml" \) -exec sed -i -e "s/0.0.0+auto.0/3.40.0/" {} +
  find -type f -not -path "./.*" -not -path "./docs*" \( -name "*.py" -o -name "*.toml" \) -exec sed -i -e "s/0.0.0+auto.0/3.40.0/" {} +
  python -m build
  twine upload dist/*
  shell: /usr/bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/[3](https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/4047615129/jobs/6961847839#step:6:3).11.1/x6[4](https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/4047615129/jobs/6961847839#step:6:4)
    TWINE_USERNAME: ***
    TWINE_PASSWORD: ***
/opt/hostedtoolcache/Python/3.11.1/x[6](https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/4047615129/jobs/6961847839#step:6:6)4/bin/python: No module named build
Error: Process completed with exit code 1.
```

makermelissa, 2023-01-30:
It is still failing. It looks like it's skipping the Install Dependencies step for some reason.

tekktrik, 2023-01-31:
I'll look into this tomorrow, I see another thing to upgrade so I'll submit a PR then

tekktrik, 2023-01-31:
Never mind, almsot positive I found the issue now

PR #278 (2023-01-31): Upgrade actions, fix output checking
Checking for the output of one of the steps was botched, fixed that.  Also took the opportunity to upgrade the CI steps as well.

Fixes #276
closed 2023-01-31 by makermelissa