adafruit/Adafruit_Python_PlatformDetect #291: Problem with Detecting  A new version of Raspberry pi 3B +

kaangoksal, 2023-05-05:
this is also not detected, I think this is a new batch of raspberry pis
```
(venv) pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

(venv) pi@raspberrypi:~ $ cat /proc/device-tree/model
Raspberry Pi 3 Model B Plus Rev 1.4(venv)
```

```
Traceback (most recent call last):
  File "/home/pi/project/src/main.py", line 7, in <module>
    from peripherals.LedController import LedController
  File "/home/pi/project/src/peripherals/LedController.py", line 2, in <module>
    import neopixel
  File "/home/pi/project/src/venv/lib/python3.9/site-packages/neopixel.py", line 16, in <module>
    import board
  File "/home/pi/project/src/venv/lib/python3.9/site-packages/board.py", line 351, in <module>
    raise NotImplementedError(
NotImplementedError: 
        Adafruit-PlatformDetect version 3.45.1 was unable to identify the board and/or
        microcontroller running the Linux platform. Please be sure you
        have the latest packages running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'
```
I played around with the detector, I think the issue is not having this version in the constants file
```
>>> detector.board._pi_rev_code()
'a020d4'
```

kaangoksal, 2023-05-05:
Here is the relevant pr 

https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/292

makermelissa, 2023-05-05:
Huh, ok. Thanks, I thought they were done updating the Pi 3.

kaangoksal, 2023-05-06:
are you going to merge it?

BradleyHazel, 2023-05-07:
Having the same issue with the same board a020d4. Thanks in advance!

makermelissa, 2023-05-08:
Oops, meant to merge sooner. Ok, merged now.

closed 2023-05-08 by makermelissa