adafruit/Adafruit_Python_PlatformDetect #296: Build failures due to board.py exceeding maximum lines permitted by Pylint

NathanY3G, 2023-05-21:
At the time of writing the `board.py` source file contains `990` lines. The maximum permitted by the project's Pylint configuration is `1000`. Attempting to add support for a new board resulted in build failures.

One possible solution to this is to refactor some of the `if-elif` constructs to use a dictionary instead.

makermelissa, 2023-05-24:
Yeah, good idea. Another solution that could help would be to move a lot of the private id functions to another module and just import from there. That way it shouldn't break anything.

NathanY3G, 2023-05-27:
That's a great idea @makermelissa. To check that I have understood your idea correctly I have created a draft PR https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/299. I have initially chosen a single but very common board to aid testing. I would be grateful if you could take a look.