adafruit/Adafruit_Python_PlatformDetect #298: Circuitpython - Generic datalogger code using platformdetect for sbc/embedded library switching

tyeth, 2023-05-23:
Hi, I know this is meant for python platforms, but the utility of it to detect if embedded or linux_sbc is immense. It needs a minor tweak to make the code compatible with circuitpython, which I've done on this branch:
https://github.com/tyeth/Adafruit_Python_PlatformDetect/tree/circuitpython-error-fix

I've done a pull request as the changes do no harm and are in line with the files syntax, being only syntactic improvements/fixes as I can see it. #297

If it makes sense to anyone else too then I'd like to have this available in circuitpython libraries, so one can `circup` your code to a working fashion in a similar vein to `pip`. I've used it for example to detect whether to switch python logging module with adafruit logging (along with a few other things related to code.py functionality).

dhalbert, 2023-05-23:
Instead of using this library, which is fairly large, you can use `sys.implementation.name`:

```py
>>> import sys
>>> sys.implementation
(name='circuitpython', version=(8, 1, 0), mpy=517)
>>> sys.implementation.name
'circuitpython'
```py
>>> import sys
>>> sys.implementation
namespace(name='cpython', cache_tag='cpython-310', version=sys.version_info(major=3, minor=10, micro=6, releaselevel='final', serial=0), hexversion=50988784, _multiarch='x86_64-linux-gnu')
>>> sys.implementation.name
'cpython'
```
There are other interesting builtins to use, such as `os.uname()`, and `sys.platform`.

tyeth, 2023-05-23:
oh god thats handy, thanks @dhalbert

closed 2023-05-23 by dhalbert