adafruit/Adafruit_Python_PlatformDetect #300: Unable to recognize boards running balena OS

jczacharia, 2023-06-14:
It looks like this issue is still ongoing. I'm using a Google Coral dev board with balena OS and running a py file with `import board` gives me this:

```Traceback (most recent call last):
File "/app/hardware-logic/api.py", line 34, in <module>
  import board
File "/usr/local/lib/python3.9/dist-packages/board.py", line 22, in <module>
    from adafruit_blinka.agnostic import board_id, detector
File "/usr/local/lib/python3.9/dist-packages/adafruit_blinka/agnostic/__init__.py", line 21, in <module>
  chip_id = detector.chip.id
File "/usr/local/lib/python3.9/dist-packages/adafruit_platformdetect/chip.py", line 419, in __getattr__
    if self.id == attr:
File "/usr/local/lib/python3.9/dist-packages/adafruit_platformdetect/chip.py", line 419, in __getattr__
    if self.id == attr:
File "/usr/local/lib/python3.9/dist-packages/adafruit_platformdetect/chip.py", line 419, in __getattr__
    if self.id == attr:
  [Previous line repeated 974 more times]
File "/usr/local/lib/python3.9/dist-packages/adafruit_platformdetect/chip.py", line 158, in id
    self._chip_id = self._linux_id()
File "/usr/local/lib/python3.9/dist-packages/adafruit_platformdetect/chip.py", line 293, in _linux_id
    hardware = self.detector.get_cpuinfo_field("Hardware")
File "/usr/local/lib/python3.9/dist-packages/adafruit_platformdetect/__init__.py", line 44, in get_cpuinfo_field
    cpuinfo = infile.read().split("\n")   RecursionError: maximum recursion depth exceeded while calling a Python object
```

I'm using these dependencies
```
pandas==1.4.1
ipython==8.4.0
smbus2==0.4.2
python-periphery==2.4.1
adafruit-blinka==8.19.0
adafruit-circuitpython-vl53l0x==3.6.9
```

I'm not sure if this is a balena issue I'm still encountering this. Any insight on a fix?

_Originally posted by @jczacharia in https://github.com/adafruit/Adafruit_Python_PlatformDetect/issues/96#issuecomment-1582972995_

makermelissa, 2023-06-15:
Balena OS support has not been added yet. It's likely the detection may need to be slightly different.

CrazyIvan359, 2023-06-17:
@jczacharia Sorry for the wait, I travel a lot for work during the week so I wasn't able to work on this. Getting back into this it seems that detection for the Coral was already there but isn't working. Can you show me the output of  `cat /proc/device-tree/model` and also `bin/detect.py` from this project?

@makermelissa It seems to be Linux based with Device Tree, based on what Jeremy has been showing me from his board though I don't know much about Balena OS so maybe that only applies to this board. Looks like support was added back in May of 2019 by Lady Ada herself in [4f0e22e](https://github.com/adafruit/Adafruit_Python_PlatformDetect/commit/4f0e22eead29b11941c505d93878648121c8896e) but it doesn't seem to be detecting the chip correctly.

jczacharia, 2023-06-17:
From `cat /proc/device-tree/model`:
```
Freescale i.MX8MQ Phanbellr
```

Detect output:
```
 Board Detection Test

 Check that the Chip and Board IDs match your board and that this it is
 correctly detecting whether or not it is a Linux board.

 Chip id:  IMX8MX
 Board id:  CORAL_EDGE_TPU_DEV

 Linux Detection
 ---------------
 Is this an embedded Linux system? True

 Raspberry Pi Boards
 -------------------
 Is this a Pi 3B+? False
 Is this a Pi 4B? False
 Is this a 40-pin Raspberry Pi? False
 Is this a Raspberry Pi Compute Module? False

 Other Boards
 -------------------
 Is this a Siemens Simatic IOT2000 Gateway? False
 Is this a 96boards board? False
 Is this a BeagleBone board? False
 Is this a Giant board? False
 Is this a Coral Dev board? True
 Is this a MaaXBoard? False
 Is this a SiFive board?  False
 Is this a PYNQ board? False
 Is this a Rock Pi board? False
 Is this a NanoPi board? False
 Is this a Khadas VIM3 board? False
 Is this a Clockwork Pi board? False
 Is this a Seeed Board? False
 Is this a UDOO board? False
 Is this an ASUS Tinker board? False
 Is this an STM32MP1 board? False
 Is this a generic Linux PC? False
 Is this an OS environment variable special case? False
 Coral device detected.
```

It looks like it's able to be detected.

CrazyIvan359, 2023-06-18:
Yeah that's what I thought... Fresh eyes see new things, yesterday looking through the code I was able to follow through detecting the chip and board based on the information you gave me. Now I'm wondering why your app is having issues. Can you post a snippet of your code where you're using Platform Detect?

jczacharia, 2023-06-18:
I was encountering the error whenever I 'import board'. I tried it by itself in a file and it errored out.

CrazyIvan359, 2023-06-19:
> I was encountering the error whenever I 'import board'.

This is probably the issue, you aren't meant to import `board` directly. See the [readme](https://github.com/adafruit/Adafruit_Python_PlatformDetect#usage-example) on the homepage of this repo for a usage example.

makermelissa, 2023-06-20:
> It looks like it's able to be detected.

Ok, that's great then.

> > I was encountering the error whenever I 'import board'.
> 
> This is probably the issue, you aren't meant to import `board` directly. See the [readme](https://github.com/adafruit/Adafruit_Python_PlatformDetect#usage-example) on the homepage of this repo for a usage example.

`import board` should work fine without issues. However, this is part of [Blinka](https://github.com/adafruit/Adafruit_Blinka) and not PlatformDetect. Despite that, based on the error messages, it does sound like the root cause is here in PlatformDetect.

Here's what I think is happening. I notice line 419 is called 3 times in a row (plus another 974 times according to the message). This is inside of the `__getattr__` function. Inside of that function, `self.id` is called which is referencing an attribute, which is probably causing the __getattr_ function to be called again. I'm kind of surprised this hasn't popped up before. I'm not sure what the correct fix for this is though.

makermelissa, 2023-06-20:
Perhaps a potential fix would be to add something like this to the beginning of `__getattr__`:
```
if attr == "id":
    return self.__getattribute__(attr)
```

CrazyIvan359, 2023-06-22:
> Perhaps a potential fix would be to add something like this to the beginning of `__getattr__`:
> 
> ```
> if attr == "id":
>     return self.__getattribute__(attr)
> ```

This is almost exactly [what I tried last time](https://github.com/adafruit/Adafruit_Python_PlatformDetect/pull/104) but found that the issue I was seeing was actually a generic x86 not being detected. That said, this recursion issue seems to happen when importing `board` directly so maybe it was 2 different issues that ended up with infinite recursion in the same place.

ChillerDragon, 2023-09-12:
Same issue on a [hetzner](https://www.hetzner.com/cloud) arm64 ampere altra machine.

The recursion error might just be a symptom but seems like something that should be fixed still.

```
root@arm-test:~/Adafruit_Python_PlatformDetect# cat /proc/device-tree/model
cat: /proc/device-tree/model: No such file or directory
root@arm-test:~/Adafruit_Python_PlatformDetect# ls /proc/device-tree
ls: cannot access '/proc/device-tree': No such file or directory
root@arm-test:~/Adafruit_Python_PlatformDetect# cat /etc/armbian-release
cat: /etc/armbian-release: No such file or directory
root@arm-test:~/Adafruit_Python_PlatformDetect# cat /proc/cpuinfo 
processor       : 0
BogoMIPS        : 50.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp ssbs
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x3
CPU part        : 0xd0c
CPU revision    : 1

processor       : 1
BogoMIPS        : 50.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp ssbs
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x3
CPU part        : 0xd0c
CPU revision    : 1

root@arm-test:~/Adafruit_Python_PlatformDetect# git log -n 1
commit 633203a7c155c43663a2b416b9d29117b4353ea6 (HEAD -> main, tag: 3.52.0, origin/main, origin/HEAD)
Merge: 8cc24b0 e4ca06b
Author: Melissa LeBlanc-Williams <melissa@adafruit.com>
Date:   Fri Sep 8 10:02:50 2023 -0700

    Merge pull request #314 from makermelissa/update-versions
    
    Update Pre-commit config versions
root@arm-test:~/Adafruit_Python_PlatformDetect# python3
Python 3.10.12 (main, Jun 11 2023, 05:26:28) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from adafruit_platformdetect import Detector
>>> detector = Detector()
>>> print("Chip id: ", detector.chip.id)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/Adafruit_Python_PlatformDetect/adafruit_platformdetect/chip.py", line 425, in __getattr__
    if self.id == attr:
  File "/root/Adafruit_Python_PlatformDetect/adafruit_platformdetect/chip.py", line 425, in __getattr__
    if self.id == attr:
  File "/root/Adafruit_Python_PlatformDetect/adafruit_platformdetect/chip.py", line 425, in __getattr__
    if self.id == attr:
  [Previous line repeated 988 more times]
  File "/root/Adafruit_Python_PlatformDetect/adafruit_platformdetect/chip.py", line 159, in id
    self._chip_id = self._linux_id()
  File "/root/Adafruit_Python_PlatformDetect/adafruit_platformdetect/chip.py", line 297, in _linux_id
    hardware = self.detector.get_cpuinfo_field("Hardware")
  File "/root/Adafruit_Python_PlatformDetect/adafruit_platformdetect/__init__.py", line 43, in get_cpuinfo_field
    with open("/proc/cpuinfo", "r", encoding="utf-8") as infile:
RecursionError: maximum recursion depth exceeded while calling a Python object
>>> 

```

PR #315 (2023-09-18): Fix recursion on some systems
Fixes #300. I found the fix for this at https://nedbatchelder.com/blog/201010/surprising_getattr_recursion.html and tested on a Raspberry Pi 4 to make sure it still works with this change.
closed 2023-09-18 by kattni