adafruit/Adafruit_Python_PlatformDetect #317: Unsupported Board (d03115)

jjklangbehn, 2023-09-18:
I am a beginner so I hope everything is understandable.

When I try to run my python script I get the following error:

> Traceback (most recent call last):
  File "/Users/user/Desktop/dev/pi/RPiFirebaseMLX90614.py", line 15, in <module>
    import board
  File "/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/board.py", line 357, in <module>
    raise NotImplementedError(
NotImplementedError: 
        Adafruit-PlatformDetect version 3.52.0 was unable to identify the board and/or
        microcontroller running the Darwin platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

> If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.

My Revision number is d03115.

Hope this is everything relevant you need.

makermelissa, 2023-09-18:
I'm moving to PlatformDetect because Raspberry Pi boards should have automatically been detected correctly with recent changes.

makermelissa, 2023-09-18:
The error mentions it is running on the **Darwin** Platform, which usually is on a Mac. Which OS do you have on your Pi? It should say "Linux".

makermelissa, 2023-09-18:
Closing because based on the script path, it appears this was not run on the Raspberry Pi itself. I have linked a guide above to help with getting up and running.

closed 2023-09-18 by makermelissa