adafruit/Adafruit_Python_PlatformDetect #332: #331 breaks microcontroller package

cybershoe, 2023-12-07:
PR  #331 removes a constant relied on by adafruit_blinka.microcontroller:

Steps to reproduce:
`import microcontroller`

```
>>> import microcontroller
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/adam/src/minihass/venv/lib/python3.10/site-packages/microcontroller/__init__.py", line 16, in <module>
    from microcontroller import pin  # pylint: disable=unused-import
  File "/home/adam/src/minihass/venv/lib/python3.10/site-packages/microcontroller/pin.py", line 117, in <module>
    elif chip_id == ap_chip.RK3568B2:
AttributeError: module 'adafruit_platformdetect.constants.chips' has no attribute 'RK3568B2'. Did you mean: 'RK3568'?
```

makermelissa, 2023-12-07:
Once https://github.com/adafruit/Adafruit_Blinka/pull/750 is updated and merged, it should work again. Unfortunately with the 2 repo system, it has to be merged in this one and released before the other one can pass the test. In rare instances like this, it can leave it in a non-working state. My suggestion would be to install [PlatformDetect 3.56.0](https://github.com/adafruit/Adafruit_Python_PlatformDetect/tree/3.56.0) until it's merged.

makermelissa, 2023-12-07:
Sounds good. Yeah, I think they just need to add a description for what the function does.

makermelissa, 2023-12-08:
The issue should be fixed now.