adafruit/Adafruit_Python_PlatformDetect #338: circuitpython-build-bundles causing CI failures

makermelissa, 2024-01-09:
@tekktrik you originally added this in #206. Is this necessary or can we remove the build bundles? Currently it is causing the following strange error in Pull Requests:

```
Run circuitpython-build-bundles --filename_prefix adafruit-python-platformdetect --library_location .
  circuitpython-build-bundles --filename_prefix adafruit-python-platformdetect --library_location .
  shell: /usr/bin/bash -e {0}
  env:
    pythonLocation: /opt/hostedtoolcache/Python/3.7.17/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.7.17/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.7.17/x64
    Python[2](https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/7440742493/job/20242085060?pr=337#step:12:2)_ROOT_DIR: /opt/hostedtoolcache/Python/[3](https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/7440742493/job/20242085060?pr=337#step:12:3).7.17/x6[4](https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/7440742493/job/20242085060?pr=337#step:12:4)
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.7.17/x[6](https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/7440742493/job/20242085060?pr=337#step:12:6)4
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.[7](https://github.com/adafruit/Adafruit_Python_PlatformDetect/actions/runs/7440742493/job/20242085060?pr=337#step:12:7).17/x64/lib
build.library failure: /home/runner/work/Adafruit_Python_PlatformDetect/Adafruit_Python_PlatformDetect
Only a single package is supported

Generating VERSIONS
WARNING: some failures above
Error: Process completed with exit code 2.
```

closed 2024-01-10 by makermelissa

tekktrik, 2024-01-10:
Just caught up on this now, I think removing was the best option.

tekktrik, 2024-01-10:
I think it was an accidental carryover.

makermelissa, 2024-01-10:
Ah, makes sense. I'm sure I've done similar.

PR #339 (2024-01-10): Remove Build Assets which is causing PRs to fail
Fixes #338. The Build Bundles is causing a failure and is not present in other non-CircuitPython repos, so it doesn't seem necessary here.