adafruit/Adafruit_Python_PlatformDetect #342: detector.chip.id  raise AttributeError()  # Avoid infinite recursion on RPI 3 B

Elwinmage, 2024-02-25:
Hi, I have an AttributeError()  trying to detect my chip id (RPI 3B)

$ pip list |grep -i adafruit

Adafruit-PlatformDetect 3.60.0

python 3.11.8 (main, Feb 19 2024, 17:01:17) [GCC 13.2.1 20231014] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from adafruit_platformdetect import Detector
>>> detector = Detector()
>>> print("Chip id: ", detector.chip.id)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.11/site-packages/adafruit_platformdetect/chip.py", line 428, in __getattr__
    raise AttributeError()  # Avoid infinite recursion
    ^^^^^^^^^^^^^^^^^^^^^^
AttributeError
>>>

$ cat /etc/os-release

NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.19.1
PRETTY_NAME="Alpine Linux v3.19"
HOME_URL="https://alpinelinux.org/"
BUG_REPORT_URL="https://gitlab.alpinelinux.org/alpine/aports/-/issues"

$ cat /proc/cpuinfo

processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Revision        : a22082
Serial          : 000000008a496a07
Model           : Raspberry Pi 3 Model B Rev 1.2

Elwinmage, 2024-02-26:
Same problem with a RPI 4B also running HAOSS (Home assistant).

makermelissa, 2024-02-26:
Interesting. Not much has changed with Blinka/PlatformDetect in a while. I wonder if this is something that is showing up in newer Python versions.

Firestorm7893, 2024-06-15:
Unfortunately I'm having the same issue, RPI 5, I am running the library through docker though.

makermelissa, 2024-06-17:
> I am running the library through docker though.

Hmm, ok. Maybe that's why I haven't seen the issue before. Thank you.

pbrownrobo, 2024-08-21:
I am having same problem, but I'm just blindly copying code from another codebase and trying to get it to work with my new hardware.

I would appreciate it if someone could spell out for idiots, what are the typical causes for this recursion problem?

torchiaf, 2024-12-30:
I had the same issue running docker on Pi5. I used `--privileged` option in `docker run` and the error went away.

pbrownrobo, 2024-12-30:
same symptom different reasons. your docker issue was simply "driver cant access hardware", but thats because you didnt give it docker privs to do so.
Mine was "driver cant access hardware" for other reasons, I believe.

BoxenOfDonuts, 2025-01-02:
Was running into a similar issue, a container that worked on a pi zero w, was throwing the same error above on a pi zero 2 w. Looks like [here](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/287633cb1bc1e1a06afcc40f32fe12e9422a80a3/adafruit_platformdetect/chip.py#L325C9-L325C63) it checks for `Hardware` in `/proc/cpuinfo`. 

On the Pi Zero it returns `BCM2835` but on the Zero 2 it returns none. It looks like the arm64 version of `/proc/cpuinfo` doesn't contain `Hardware`.

When not running in a container the chip will be detected via information in `/proc/device-tree/compatible` [here](https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/287633cb1bc1e1a06afcc40f32fe12e9422a80a3/adafruit_platformdetect/chip.py#L437), but that doesn't exist inside the container without setting `--privileged`.

I was able to get around it by setting the container to run privileged, or passing `BLINKA_FORCEBOARD` and `BLINKA_FORCECHIP` environment variables

harshil21, 2025-03-01:
Getting the same error in arm64 github action runners. 

> passing BLINKA_FORCEBOARD and BLINKA_FORCECHIP environment variables

what exactly should I set it to? I just want a mock interface, I'm not trying to control something remotely.

BoxenOfDonuts, 2025-03-01:
>what exactly should I set it to? I just want a mock interface, I'm not trying to control something remotely.

For the pi zero 2w I set it to these. FYI I ended up just setting the container to run as privileged. 
```
BLINKA_FORCEBOARD=RASPBERRY_PI_ZERO_2_W
BLINKA_FORCECHIP=BCM2XXX
```