adafruit/Adafruit_Python_PlatformDetect #349: Problem with vim3 board -Linux Ubuntu 4.9

HudaHussamEldeen, 2024-04-18:
### Board Name

Khadas vim3

### Steps

when i include any package from adafruit ..it doesn't able to identify the board 
i'm up to date with libraries and adafruit supports vim3 board
so i don't know the reason of this problem 

### Description

Traceback (most recent call last):
  File "/home/khadas/Desktop/intilligent module/adafruit_can.py", line 1, in <module>
    import board
  File "/home/khadas/.local/lib/python3.8/site-packages/board.py", line 405, in <module>
    raise NotImplementedError(
NotImplementedError: 
        Adafruit-PlatformDetect version 3.62.0 was unable to identify the board and/or
        microcontroller running the Linux platform. Please be sure you
        have the latest packages by running:
        'pip3 install --upgrade adafruit-blinka adafruit-platformdetect'

        If you are running the latest package, your board may not yet be supported. Please
        open a New Issue on GitHub at https://github.com/adafruit/Adafruit_Blinka/issues and
        select New Board Request.

makermelissa, 2024-04-18:
What are the results of typing `cat /proc/device-tree/compatible` in the terminal?

makermelissa, 2025-08-27:
Closing due to lack of information.

closed 2025-08-27 by makermelissa