adafruit/Adafruit_Python_PlatformDetect #352: orangepipc2 detection is wrong

nopnop2002, 2024-05-29:
For orangepipc2:
```
board_value = self.detector.get_armbian_release_field("BOARD")
print("_armbian_id board_value={}".format(board_value))
```

_armbian_id board_value=orangepipc2

```
elif board_value == "orangepi2":
    board = boards.ORANGE_PI_2
```

The naming is:
Orangepi -> Orangepi 2 -> Orangepi PC -> Orangepi PC2 -> Orangepi 3 -> Orangepi 4 -> Orangepi 5.
Before Orangepi PC was released (around 2015), there was a model called Orangepi 2.

makermelissa, 2024-05-30:
Makes sense. Would you like to submit a PR to update it?

nopnop2002, 2024-05-30:
Adafruit_Python_PlatformDetect and Adafruit_Blinka must be changed at the same time.

Therefore, I think it is better to leave boards.ORANGE_PI_2 and add boards.ORANGE_PI_PC2.

nopnop2002, 2024-06-13:
I added orangepi pc2.

closed 2024-06-13 by nopnop2002